bigtreetech/BIGTREETECH-SKR-mini-E3 #407: PartCooling FAN0 Suddenly Always at 100%

redpiyaya, 2020-09-24:
Using BTT SKR MINI E3 V2.0 + BTT TFT35 V3.0 on Ender 5 Plus
Was working for a few days, no issue. Suddenly Part Cooling FAN0 is on at 100% all the time and cannot be controlled by screen, BTT Mode or Marlin Mode or even Octopi. Reset EEPROM, Restore defaults, Reflashed same and different firmware = SAME.
I wonder why It would suddenly do that. Please help.

trunglp, 2020-10-01:
I same problem on board SKR mini E3 v1.2 . Fan alway is 100%,  I replace  mosfet on board, it is work.

evilpoly, 2020-10-01:
witch mosfet did you replace ? i think i have to do the same

trunglp, 2020-10-02:
> 
> 
> witch mosfet did you replace ? i think i have to do the same

you can use SOT-23

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/238#issuecomment-633108047