bigtreetech/BIGTREETECH-SKR-mini-E3 #409: LCD doesn't power up after installing BL Touch

BannanaFarmer, 2020-09-26:
I installed an SKR Mini V1.2 recently and tried installing a bl touch as well. I got the Creality BL Touch kit and I can't figure out why the LCD doesn't turn on when I power the machine on. I plugged in all five wires correctly and updated the firmware yet the LCD doesn't power on. The only way it will power on is if I plug the three-pin connector a slot over (PT-DET as labeled on the board), but when I try testing the probe it doesn't do anything. If I try to home it the x-axis would crash into the build plate. I have no idea what to try next.

db250336, 2020-10-05:
Hi,
If your sensor is  3dtouch you need to chane positions in the 3 pin cable that connect to the skr mini e3

BannanaFarmer, 2020-10-05:
> Hi,
> If your sensor is 3dtouch you need to chane positions in the 3 pin cable that connect to the skr mini e3

BannanaFarmer, 2020-10-05:
My sensors a genuine BLTouch, I have changed the position of the 3 pin connector but the problem still persists

marcosscriven, 2020-10-20:
I'm having the same issue @BannanaFarmer - did you resolve this?

BannanaFarmer, 2020-10-21:
I haven't @marcosscriven. I'm thinking of uninstalling the bl touch altogether and calling it quits.

marcosscriven, 2020-10-21:
In my case it was due to the GND and 5V being flipped in the wiring from the BLtouch.

DarkPand0r, 2020-11-15:
@marcosscriven is right, I had the same issue with mine. Looked correct until I noticed the cables were flipped as they went into the board connector. Thanks for the tip mate!