bigtreetech/BIGTREETECH-SKR-mini-E3 #410: Version 2.0.6.1 Z Homing does not work

Mechtech63, 2020-09-28:
I have flashed the firmware for Z homing version 2.0.6.1 and the Z homing does not work.  All BL Touch menus that should be there are gone.  As a verifier I flashed the regular version 2.0.6.1 without Z homing and the BL Touch menus are back, except auto homing with this results the Z crashing through your print bed.

### Steps to reproduce

1. Download latest file added 9/21/20 with Z Homing
2. Flash to Ender 3 with Bigtreetech--SKR installed with a BL-Touch version 3.1 attached.
3. Go to configure settings and look for BL Touch menu.
4. Repeat steps, but download the non Z Homing version.
5. Check configuration to see BL Touch menus return.

**Expected behavior**
After flashing the firmware, you should be able to go to "Motion" and select "Auto home".
The BL Touch probe will center over the print bed and extend the probe tip to probe for the print bed.
The extruder will move downward in Z and stop when the probe tip touches the print bed and gets you close.
You make final adjustments by adjusting the Z offset.

**Actual behavior**
No BL Touch menus are present after flashing firmware, so way to start the homing of the Z axis.

### Additional Information
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V1.2/firmware-bltouch-for-z-homing.bin

BramEsendam, 2020-09-28:
I'm having the some problem.

AEnoch29, 2020-09-28:
Having the same problem as well

Slimgig, 2020-09-28:
I had the same problem as well I fix it but dont remember what I changed let me look in to it real quick and see if I can find it

Slimgig, 2020-09-28:
Did you enable #define z_min_probe_uses_z_min_endstop_pin

Slimgig, 2020-09-28:
And is your probe connected to the z-min pin on the board the black and white wire ?

AEnoch29, 2020-09-28:
> 
> 
> Did you enable #define z_min_probe_uses_z_min_endstop_pin

I followed the guide that instructed to just drag and drop the firmware-bltouch-for-z-homing.bin file to the SD card and rename it to firmware.bin

AEnoch29, 2020-09-28:
> 
> 
> And is your probe connected to the z-min pin on the board the black and white wire ?

Yes, it matches the diagram found here

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2

Mechtech63, 2020-09-28:
> And is your probe connected to the z-min pin on the board the black and white wire ?

Yes, it is.

Mechtech63, 2020-09-28:
> Did you enable #define z_min_probe_uses_z_min_endstop_pin

I took the path of not having to edit the code since you don't have to with the SKR mini board.  It's this latest version of the firmware that got some stuff left out.

abacabie, 2020-09-29:
I think to get the best out of this board it will certainly be worth the effort of learning to build The firmware  yourself, it may look daunting but it’s nowhere near as difficult as it seems and there are plenty guides and videos to help, it is certainly worth the effort.

BramEsendam, 2020-09-29:
> I think to get the best out of this board it will certainly be worth the effort of learning to build The firmware yourself, it may look daunting but it’s nowhere near as difficult as it seems and there are plenty guides and videos to help, it is certainly worth the effort.

I agree, but the correct pre compiled firmware should still be available. Some features are just not enabled/working right now for `firmware-bltouch-for-z-homing.bin`, `firmware-bltouch.bin`.

local3k, 2020-10-02:
Can anyone share ur workin firmware with z homing?

abacabie, 2020-10-02:
> Can anyone share ur workin firmware with z homing?

if you confirm what board you have and it is indeed for the ender 3/pro i'm pretty sure someone will help.

ayman0201, 2020-10-03:
Same issue with ender 3 pro. The stepper motors are also very loud (almost grinding sound). All hardware works perfectly with older firmware I have.

ta4chu, 2020-10-03:
Hi Guys; If possible can anybody share the fixed '' firmware-bltouch-for-z-homing.bin '' for  SKR Mini V1.2 mainboard for Ender 3 Pro? Thanks for your help.

Mechtech63, 2020-10-03:
I found this firmware version and it works perfectly at Z homing.  Does 7 x 7 leveling. It's perfect for my Ender 3 with SKR min E3 1.2.Https://github.com/gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2/blob/master/CompiledFirmware/firmware.bin

ta4chu, 2020-10-03:
Hi; Thanks for your help. I cannot access the file in the link you shared. Could you share again?
[image]

Mechtech63, 2020-10-03:
Here is a photo of the link I followed.

ta4chu, 2020-10-03:
I guess I'm doing something wrong. In your last post, "Here is a photo of the link I followed." Which photo I can not see. If possible, would you upload and share the file on a link.

Mechtech63, 2020-10-03:
I'll do that once home.
Also. You can go to github and try to find the links that were in my post.  Maybe you'll get lucky.

Mechtech63, 2020-10-04:
I have attached the firmware file to this email.

KapiteinHaddock, 2020-10-07:
I found it using this link! Just search for the user on github and you will find the firmware.
https://github.com/gazcbm/Marlin-2.0.x-SKR-Mini-E3-v1.2/blob/master/CompiledFirmWare/firmware.bin

asprada, 2020-10-08:
The files you provided are very old builds. I used the file from teachingtech but I've got problem that every time I auto home the Z distance to bed is different. During printing I can feel that the Z motor is rotating to correct the bad bed leveling but the print is not ok (it looks like the 3d touch was not working). Any ideas?

JPMATIZ, 2020-10-28:
I Installed an BTT SKR MINI E3 V2.0 and the BL on a Ender 3 Pro but when I try to do a Z Homing the Z axis is far apart from the bed and the BLT activate 3 times , and then appear a M999 BLtouch error . So I check the wiring and its good . I Uploaded the "firmware-bltouch-for-z-homing.bin"  and the problem still appear . So I Configured the Marlin again and uploaded but the error appear again . Any ideas what could be the  problem ?