bigtreetech/BIGTREETECH-SKR-mini-E3 #412: Release 2.0.6.1 (with Z homing) FAN 1 not triggering when stepper drivers activated

Silversunbeam, 2020-09-28:
When engaging any of  the stepper motors (i.e., X axis, Y axis, Z axis, Extruder) the fan remains off. 

**Expected behavior**
Fan1 (control board cooling fan) should run when any of the stepper drivers are engaged and then return to idle once the stepper activity time-out period has been reached.

**Actual behavior**
Fan1 remains off.

Previous firmware release operated normally

Marco-Santos20, 2020-09-30:
Good afternoon my name is Marco Santos I have an SKR MINI E3 V2.0 bord I have all the cables well fitted in the right quote but the fan where the SKR MINI E3 V 2.0 bord is not working I would like what I can do for it to work

Boa tarde meu nome é Marco Santos eu tenho uma bord SKR MINI E3 V2.0 eu tenho os cabos todos bem encaixados no cito certo mas a ventoinha onde esta a bord SKR MINI E3 V 2.0 não esta a funcionar eu gostaria o que eu possa fazer para ela funcionar 
o sr conseguiu resolver o problema ?

Silversunbeam, 2020-10-01:
> Good afternoon my name is Marco Santos I have an SKR MINI E3 V2.0 bord I have all the cables well fitted in the right quote but the fan where the SKR MINI E3 V 2.0 bord is not working I would like [...]

My solution was to downgrade to an earlier version of the firmware.

BalanBro, 2020-10-01:
Mine is doing the same with the current Z homing firmware.

BalanBro, 2020-10-07:
I too downgraded to the firmware from May 26th with BL Touch Z homing (I think), and all seems to be well with the fans now.  This particular firmware was also set up for 25 point ABL vs. 9 point on the latest version, which is something I wanted anyway. I guess this ended up being a blessing in disguise, lol.

eravau, 2020-10-14:
Same issue with firmware.bin of Jun 2, 2020. Fan1 (motherboard cooling fan) remains off. 
I reverted to firmware.bin of May 21, 2020 in commit 8962900 and Fan 1 rotates on stepper movement as expected.

Marco-Santos20, 2020-10-14:
good morning but just download the may 21 firmwer and put it on ender 3 pro or do i have to use the firmwer?

bom dia mas é so fazer o download do firmwer de 21 de maio e por na ender 3 pro ou tenho que mecher no firmwer ?

BalanBro, 2020-10-14:
> good morning but just download the may 21 firmwer and put it on ender 3 pro or do i have to use the firmwer? bom dia mas é so fazer o download do firmwer de 21 de maio e por na ender 3 pro ou tenho que mecher no firmwer ?
> […](#)

I'm not certain of what your question is, but if you have the SKR Mini E V2.0 board, all you have to do is upload the older firmware via microSD.  Just make sure you choose the right configuration for you model (Standard, BLTouch or BLTouch with Z homing).

geladk, 2020-10-14:
Where can I find earlier firmware releases?  I’m running E3 V2 with BLTouch for z-homing.

timqui, 2020-10-15:
I can confirm same problem - Ender 3 + SKR mini E3 v2 = firmware.bin from 22 Sep 2020 makes the cpu/board fan always off.

altadeos, 2020-10-16:
Same problem here with the latest firmware released today.

Silversunbeam, 2020-10-17:
There's still a bug in the config plus an added bonus regarding the z-stepper. Fan 1 comes on once the drivers are activated but then times out at some point and stops running. I think there's also a glitch in the z-stepper current setting now as well, as the motor is much quieter but seems to slip a little bit and causes the nozzle to impact slightly on the print in progress. Once again I rolled back to the Marlin bugfix release from May and everything appears to work normally.

altadeos, 2020-10-17:
> There's still a bug in the config plus an added bonus regarding the z-stepper. Fan 1 comes on once the drivers are activated but then times out at some point and stops running. I think there's also [...]

Hello, 
Could you please tell me where to find May release ? 
Best regards,

Silversunbeam, 2020-10-18:
You have to work your way back through the history. I think this link is correct:

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/1fb6a6ab3118a5f6c056d7697a18d3debfc9518f/firmware/V2.0

eravau, 2020-10-18:
Seems solved for me in firmware.bin with Marlin 2.0.7.2. Steppers come up when motor is activated.

Silversunbeam, 2020-10-27:
> Seems solved for me in firmware.bin with Marlin 2.0.7.2. Steppers come up when motor is activated.

That seemed to be the case for me too, initially. I found that when I went to print anything other than a single layer calibration square Fan1 (the control board fan) once again shut down and did not restart.

mcarter14777, 2020-11-10:
See #448 
Also check out the [marlin speaker docs ](https://marlinfw.org/docs/configuration/configuration.html#speaker)

I rolled my own firmware with 2.0.7.2 as I wanted to change some things and the controller fan wouldn't come on until I clicked the knob. Out of the box my skr mini e3 v2 with 2.0.x firmware worked as it should, but on the new firmware the controller fan wouldn't come on when steppers or heaters activate.

Seems like the new firmware provided here enables the speaker option by default while it was disabled in 2.0.x, so I disabled the speaker option and now the controller fan works automatically as it should. This change doesn't disable sounds for the LCD.