bigtreetech/BIGTREETECH-SKR-mini-E3 #413: SKR MINI E3 V2.0+TFT35 E3 v3.0 firmware instructions HELP

ScrapyCocos, 2020-09-29:
Hello all,

I'm complete new to 3d printing and also to code and firmware's. I use a Creality Ender 3 pro.

I do want to apologize before for any inconvenience that I'm causing asking this question, but i searched and watched 100 videos and went over at least 100 web pages before. I spent my Saturday evening/night until 5 o'clock in the morning and the whole yesterday evening to look across multiple locations( various videos and web pages) to get bits and bobs about : what software to use for modifying the firmware code(Visual Studio Code, PlatformIo,Auto Build Marlin,...)(what to activate and what not to activate in the configurationH file/configurationADV, information which you need to filter because some is old and not up to date or the videos that you find either are skipping some info either they are not using the BTT TFT35 screen and after you are done you find out that you have missing/ not working buttons in the touch version mode),steps to follow to load everything without having errors(everybody is got their way of doing this and is really confusing what to do and not to do), modifying the firmware to suit what you got (I couldn't find one tutorial to follow, which will have all steps for what to modify in your code - in my view should be something like one tutorial under each version of board with bltouch active/not active; with orig screen/btt tft35/50/75 screen).

This is just a small description  with what I went through and why i'm so frustrated with this upgrade which in my eyes should be a lot more accessible to everybody regardless of they knowledge level. From this will benefit both sides BTT and new customers.

I still couldn't made my printer to work fully with those new upgrades.

My question is: could anyone help me and the other guys that are in the same situation, with step by step guide with what to load(what order and what extensions to use in Visual Studio code so there will be no problems), the code that needs to be activated/not activated in the platform.ini, conficurationH, configurationADV to accommodate the BTT SKR mini E3 v2.0 and BTT TFT35 E3 v3.0 without the bltouch, with mesh bed leveling feature active (also while talking about mesh leveling feature, is there any way to change the stored mesh bed leveling figures after you saved them to EEporm if you want to do slide adjustments), what files to load on the SD card to do the firmware update.  

Thank you in advance!

empakoso, 2020-09-30:
3D Printing

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/413

It may seem very overwhelming at first.  The only way to get around this is to read, search and ask specific questions.  BTT has everything here in their repository.  Let’s start here:

1. The firmware w/o BLTouch that will work out of the box for you has already been I stalled on your card.  It if you’d like to reflash it to start fresh you can download it here: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware.bin
2. You can also download this entire link to customize your settings https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0 using GitHub Desktop and begin modifying using VScode since it is the base configuration for your SKRmini E3 V2.0 and TFT35 E3 V3.0 but without UBL enabled. 
3. VScode is installed using this document to start: https://github.com/bigtreetech/Document
4. To enable UBL for your bed leveling and your eeprom mesh saving question you must read this first (unified bed leveling documentation link found here): https://marlinfw.org/docs/gcode/G029-ubl.html. It is everything you every wanted and needed in multi-mesh management and multi-point mesh settings grid.
5. To change your firmware it’s simply, compile VScode, get the firmware.bin file you make with this compile in the pio sub folders. Then copy to a microSD card of 8gb or less.  Power down your SKR board and insert the sdcard.  It will update with all your changes.
6. If you have questions about your Marlinfw settings setup you must reference the following: https://marlinfw.org/docs/basics/install.html (configuration pages and/or GitHub pages for information)
7. Remember that you TFT35 E3 V3.0 runs normally using the “reprap graphical lcd” exactly the way BTT has set it up in the above link in step 2.0.
8. Great link on how to wire everything on your/my ender3 pro is here https://imgur.com/a/idnV4q6 from this awesome guide: https://youtu.be/mtCz_-2zvZo

I recommend you start with just enabling UBL and using the default settings for now.  Should be enough.  Enable G26 mesh test in there as well to help.

Good luck.

MrSmith9, 2020-10-10:
Sorry for my English ( real life. i am deaf ) i have ender 3 with SKR mini E3 v2 with TFT35 E3 V3 and my ender full working with firmware. I have been upload firmware for tft35. Ender with 2.0.7bugfix

P.S I have tired new version of TFT35 keep errors on my tft35 so I have downgrade version for TFT35 with full working. hope that help. I will upload full source. if i got time. 

https://github.com/MrSmith9

pointblack87, 2021-12-31:
Hello good afternoon.
Someone has the same information given.  https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/413#issuecomment-701373001 in the previous post, only for the SKR MINI E3 V3.0 motherboard.
since I use an optical limit switch. which has to modify a line in the marlin to make it work, but as this board is new I can't find a kernel to be able to modify and then compile.

runkster428, 2022-01-04:
Try this configuration guide - https://www.makenprint.uk/3d-printing/3d-firmware-guides/marlin/consolidated-marlin-guide/

This lays out any changes that need to be made to the firmware setting step by step.  I used it to setup my ender 5 pro and used the ender 3 printer selection option on the site as most of the settings are similar.  The only problem I had was that the controls were reversed on my TFT35 display which I was able to fix by changing the TFT35 configuration (not in the marlin firmware).

They also have a board install/wiring guide here: https://www.makenprint.uk/3d-printing/3d-printing-guides/3d-printer-mainboard-installation-guides/btt-skr-mini-e3-v3-guides/btt-skr-mini-e3-v3-setup-guide/