bigtreetech/BIGTREETECH-SKR-mini-E3 #414: CHOPPER Timing on TMC2209 with Ender 3/3pro/5 setting at 12V instead of 24V

thrutheframe, 2020-09-29:
In configuration_adv.h file. I notice that the chopper timing is at 12V.

#define CHOPPER_TIMING CHOPPER_DEFAULT_12V

Should it not be at 24V as Ender 3/3pro/5 all run on 24V PSU?

#define CHOPPER_TIMING CHOPPER_DEFAULT_24V

Please advice.

abacabie, 2020-09-30:
Marlin 2.0.7 is out now and the config for the Ender 3 pro is #define CHOPPER_TIMING CHOPPER_DEFAULT_24V

thisiskeithb, 2020-10-08:
It seems that BigTreeTech pulled a default config and lightly changed it instead of using the previously correct configs. 

I've put in a PR to update the configs in this repo (https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/pull/421), but I'd recommend pulling the configs from the [main Marlin Configurations repo](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3) instead as they are already good to go for `2.0.7`.

thrutheframe, 2020-10-10:
thank you very 

> It seems that BigTreeTech pulled a default config and lightly changed it instead of using the previously correct configs.
>
> I've put in a PR to update the configs in this repo (#421), but I'd [...]

thank you very much.

closed 2020-10-10 by thrutheframe