bigtreetech/BIGTREETECH-SKR-mini-E3 #419: Please help test Marlin firmware config for BigTreeTech SKR Mini E3 2.0 on stock Creality CR-10 and CR-10S

Gamester17, 2020-10-01:
**Is your feature request related to a problem? Please describe.**

Please help me test/verify these examples Marlin configuration files for CR-10 and CR-10S that is submitted for upstream inclusion:

https://github.com/MarlinFirmware/Configurations/pull/237

and

https://github.com/MarlinFirmware/Configurations/pull/238

Those configuration files should work with stock CR-10 and CR-10S only upgraded with the BigTreeTech SKR Mini E3 2.0 board.

**Describe the solution you'd like**

There are no config example files for BTT SKR Mini E3 2.x with presets for Creality CR-10 and CR-10S, as per request:

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/309

There are however already BTT SKR Mini E3 2.0 example config files for Ender-3, Ender-3 Pro, and Ender-5 here:

https://github.com/MarlinFirmware/Configurations

I would like for upstream Marlin Firmware and BigTreeTech/Biqu to also provide recommended presets for CR-10 and CR-10S.

**Describe alternatives you've considered**

The alternative is for all end-users to have to figure out their presets even if they have a stock CR-10 and CR-10S 3D-printer.

The learning curve for Marlin Firmware optimization is relatively steep to get started with and it would be good if customers owning CR-10 and CR-10S who buy your SKR Mini E3 2.0 could get it going more quickly with config files and binaries already available as long as you have a stock 3D-printer from Creality with no hardware modifications as of yet.

**Additional context**

In my humble opinion, having example files for specific stock printer models make it less intimidating for beginners who own a stock 3D-printer to upgrade to example BigTreeTech SKR Mini E3 2.0 board.

Best would also be if @bigtreetech could provide pre-configured and pre-compiled binary files for Creality CR-10 and CR-10S as already available for Creality Ender 3 and Ender 5 as there are many who like to upgrade their stock printers without compiling their own firmware:

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V2.0

Please also see feature requests https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/309 and https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/295 which asks for config files and binaries with pre-configured presets for CR10 and CR10S.

Gamester17, 2020-10-05:
FYI, MarlinFirmware Configurations upstream have now committed BigTreeTech SKR Mini E3 2.0 config examples for both stock Creality CR-10 and stock Creality CR-10S

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10/BigTreeTech%20SKR%20Mini%20E3%202.0

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10S/BigTreeTech%20SKR%20Mini%20E3%202.0

Now all that is missing is for @bigtreetech to provide precompiled SKR Mini E3 2.0 binary firmware files for CR10 and CR10S .

FlyBoyRez, 2020-10-06:
Many thanks for the code. I'm using this as the basis for my CR-10S4 with the SKR Mini E3 v2.0 board and the printer looks to be working well, as in it works and all directions are good.

Currently struggling with a BLTouch issue, once that's sorted can give a definitive response on how well the printer prints with the new control board.

vazsingh, 2020-11-14:
Hello @Gamester17 

I am rebuilding my CR10 that has a BTT SKR Mini E3 V2 mainboard installed and would be more than happy to test this later this week. However, I am not confident with compiling firmware. Would you be able to provide a link to a precompiled .bin that I could download?

Also, are you able to include a version with the BL touch enabled? 

Many thanks

slmalb, 2020-11-15:
I used  @Gamester17 's 10S config files and modified  for CR-10 S5 
 this board  is working nice  with CR-10 S5 printer with Bltouch. 
Just whatever I do I cannot make it work for power loss resume feature.

lajo-osd, 2020-12-14:
> Thanks for the code . I used @Gamester17 's 10S config files and modified for CR-10 S5
> this board is working nice with CR-10 S5 printer with Bltouch.
> 10S is the base of all 10 S4 and 10 S5 printers. Just jerk, bed and some small changes made it work nearly perfect.
> Just whatever I do I cannot make it work for power loss resume feature. 10S with skr mini e3 v2.0 users has a problem with power loss resume?

I have a friend that has issues getting his CR-10 S5 working with this board and the new screen, no Bltouch though.
Could you possibly share a working firmware file, configuration files or the changes made to the configuration files please?

Thanks!

Hedda, 2020-12-14:
Not got CR-10 S5 to test but there are config examples for most Creality printers the in an upstream MarlinFirmware repository:

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality

For the CR-10 S5 you should be able to use examples for CR-10 (Single Z Lead) or CR-10S (Dual Z Lead) with changed bed-size.

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10/BigTreeTech%20SKR%20Mini%20E3%202.0

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10S/BigTreeTech%20SKR%20Mini%20E3%202.0

So re-mix the CR-10 S5 example for CrealityV1 with one of those.

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10%20S5/CrealityV1

Suggest that you test that and submit any suggested improvements as a pull request to that same repository.

lajo-osd, 2020-12-14:
> Not got CR-10 S5 to test but there are config examples for most Creality printers the in an upstream MarlinFirmware repository: [...]

I'll give that a try, so only bed-size changes to configuration.h should be needed you think to get it working?

Hedda, 2020-12-14:
Bed size and Z_MAX_POS corresponding to endstop position I believe, but it is a bigger bed so maybe heating too? 

Again, suggest comparing with CrealityV1 board for CR-10 S5

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10%20S5/CrealityV1

slmalb, 2020-12-14:
> I have a friend that has issues getting his CR-10 S5 working with this board and the new screen, no Bltouch though.
> Could you possibly share a working firmware file, configuration files or the [...]

Using SKR mini E3 v2.0 ? and
Whats your/your friend's add-on(s) ? (name of the new screen?)
I have some upgrades like silicone heatbed, tft35 e3 v3.0 screen, need to edit for you maybe.

lajo-osd, 2020-12-15:
Thanks a million @slmalb !

I have compiled a new firmware based on your settings for him to test.

I'll do my best to answer your questions and I also encouraged him to register here and reply himself:

**- Whats your/your friend's issue?**
He has some issue on the Y axis, if he hits the breaker it stops as expected but if he push it quickly it stops and goes back a little and produces an error code:
_Homing Failed:
PRINTER HALTED
Please Reset_

He thinks it could be related to speed settings or bump settings.
I'm extremely new to all of this myself :)

**-Using SKR mini E3 v2.0 ? and 12V power supply?**
Yes that is correct.

**-Your add-on(s) ? (name of the new screen?)**
BIGTREETECH TFT35 E3 V3.0, this one: https://www.bigtree-tech.com/products/bigtreetech-tft35-v3-0-touch-screen-display-marlin-and-tft-touch-mode-smart-controller-panel-for-skr-v1-4-skr-pro-skr-mini-board-of-3d-printer-part.html

slmalb, 2020-12-15:
The hotbed is huge and heavy Y axis accerelation must be lower to stable movement  of  Y axis of Cr-10 S5

For this issue lowering accerelation of Y axis may solve it.

original was 500 500 100 1000 , Y axis was banging to the Z=0  point real hard Z value may be 300-350 slower is better. 
 500 was high accerelation so reduced. X maybe still 500. E is decreased littlebit.
I am happy with these values as I mentioned before.
```
#define DEFAULT_MAX_ACCELERATION      { 400, 300, 100, 850 } 
```
So waiting for the results. @lajo-osd 

 And dont know what to do or how to add a SKR mini E3 config file for CR-10S5  that may everyone directly download from config files. I someone give a hand may others can use in the future.

lajo-osd, 2020-12-16:
@slmalb My friend has reverted to the stock setup to ensure the breaker and the Y axis movement is working as expected and then will will switch boards again. 
I'll update again on he is back on the SKR board around the progress.

Thanks for the help so far!

Fluke816, 2021-01-11:
I have just installed the skr mini V2 in my CR-10, along with 24v powers supply, BLTouch, 24v fans and the TFT35 E3 V3.0 Touch Screen Control Board. I have tried building new firmware using the latest Marlin and the Configurations-release-2.0.7.2, but keep coming up with errors when I try to build the firmware with VSC. I am using the STM32F103RC_environment and have also tried the 512k version of the same, still no look.

Has anyone got firmware that would work with my new build? Thanks

Hedda, 2021-01-14:
Anyone here tested or the new "**SKR E3 Turbo**" board from BTT/Biqu on stock Creality CR-10 or stock Creality CR-10S? 

https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo

https://www.biqu.equipment/collections/hot-deals/products/bigtreetech-skr-e3-turbo-32-bit-control-board-integrated-tmc2209-uart-for-ender-3

https://www.bigtree-tech.com/products/btt-skr-e3-turbo.html

https://github.com/bigtreetech/BIGTREETECH-SKR-E3-Turbo

https://github.com/MarlinFirmware/Configurations/pull/223

https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/CR-10S/BigTreeTech%20SKR%201.4%20Turbo%20TMC2209

https://github.com/MarlinFirmware/Marlin/pull/19500

https://github.com/MarlinFirmware/Marlin/pull/19513

# Description

BigTreeTech's SKR E3 Turbo is the next board in the SKR E3 series, but with a LPC1769 MCU and a slightly larger footprint for dual extruders/hotends or independent dual Z depending on firmware configuration.

## Board details:

- 120MHz LPC1769 MCU
- 5 x integrated TMC2209s (TMC2209 s) with Trinamic's Low Power Standby feature* and sensorless homing
- Onboard EEPROM
- 4-layer PCB
- USB port ESD protection via USBLC6-2 diode
- Onboard dual z connectors when using a single Z driver or independent dual Z when repurposing the second extruder driver
- Neopixel support (without needing to mod Marlin/change to third-party libraries)
- TFT support
- Dedicated probe port
- Powerloss detection via external module
- Dual, independently controlled, fan headers
- Power supply control via external module
- Dual extruder filament runout detection
- Onboard thermistor (currently disabled as Marlin doesn't have a "board sensor" feature)

* There will be a separate PR to support Trinamic's Low Power Standby feature, but there are some essential changes in this PR to a couple core Marlin files in the interim, otherwise users will get a TMC CONNECTION ERROR on boot.

I've also put in a PR for the base config (single extruder setup), 2-into-1, and dual Z in the Marlin Configurations repo. Link below.

# Configurations

See the matching config PR in the Marlin Configurations repo: MarlinFirmware/Configurations#223