bigtreetech/BIGTREETECH-SKR-mini-E3 #420: [BUG] V2.0 firmware for ender 3 pro without BLTOUCH has too high of stepper current.

aaronsta1, 2020-10-07:
### Description

it seems the default stepper current is set to 800 for all motors.

### Steps to reproduce

just go to configuration, advanced, TM Drivers, current and see they are all set to 800.

**Expected behavior**
the motor to stay nice and cool

**Actual behavior**
motors get molten hot.

### Additional Information
the newest pre made firmware.bin file from the 2.0 github.

thisiskeithb, 2020-10-08:
It seems that BigTreeTech pulled a default config and lightly changed it instead of using the previously correct configs. 

I've put in a PR to update the configs in this repo (https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/pull/421), but I'd recommend pulling the configs from the [main Marlin Configurations repo](https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3) instead as they are already good to go for `2.0.7`.

aaronsta1, 2020-10-08:
i just tried to compile my own firmware with those config files.
do i choose the btt_512k version? 
also it wont compile, it says 
#error "EMERGENCY_PARSER is not yet implemented for STM32F1. Disable EMERGENCY_PARSER to continue."

also the master is 2.0.6.1 where do i get 2.0.7?

thisiskeithb, 2020-10-08:
Sounds like you're using an old install. Make sure you download Marlin 2.0.7 as `EMERGENCY_PARSER` wasn't implemented before that.

It may be easier if you download from the main Marlin website instead as it has direct links for both https://marlinfw.org/meta/download/

aaronsta1, 2020-10-08:
hmm, thanks ill do that one.
i just went here
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3
hit the master branch and hit the download zip.
i guess the whole thing needs updated?

thisiskeithb, 2020-10-08:
This repo (the one you’re commenting in) doesn’t have the latest version of Marlin.

The link I posted above will be the latest copy of Marlin + tested SKR Mini configs.

aaronsta1, 2020-10-08:
it worked, got 2.0.7 installed and the steppers are right.. thanks so much!

closed 2020-10-08 by aaronsta1