bigtreetech/BIGTREETECH-SKR-mini-E3 #422: Issue with endstops

LegendArmour, 2020-10-08:
Issue: X and Y endstops will not trigger individually. In order for and endstop to be triggered, the opposing endstop must be triggered. Z endstop is triggered by probe and works as intended.

Printer: Ender 5 Pro Hasn’t even been used yet

Steps to reproduce: I honestly don’t know the printer hasn’t even been used

What I’ve tried: I have tried inverting the endstop logic no difference, possibly a pin issue I really don’t know. I have tried with an endstop from my Ender 3 and the problem still persists. I’ve tried all the pre-compiled firmware’s from the git, all have the same problem with the added problem of the steppers are inverted.

LegendArmour, 2020-10-08:
Bit of an update, swapped the board into my Ender 3 and used both the pre-compiled and factory board firmware and both have the same issue, so this tells me that there is a problem with the board itself. I’m handy with electronics and smd work so does anyone know what might cause this problem maybe I can rectify it

bri-s, 2021-02-04:
I have same problem. The board is not responding to either Y and X endstop. But Z endstop works fine. 

I am using the precompiled firmware ver: 2.0.7.2 on the MINI E3 V1.2
This is the result from M119:

(when endstop X is activated) 
SENT: M119
READ: Reporting endstop status
READ: x_min: TRIGGERED
READ: y_min: open
READ: z_min: open
READ: ok    

(when endstop Y is activated) 
SENT: M119
READ: Reporting endstop status
READ: x_min: open
READ: y_min: TRIGGERED
READ: z_min: open
READ: ok

stewe93, 2021-09-17:
I have the same problem guys, do you have any solution since?

redpeppers-01, 2024-09-09:
exact same issue here also, m119 verifies the endstops are working but the axis doesnt stop movements when endstop is triggered.