bigtreetech/BIGTREETECH-SKR-mini-E3 #424: Will SKR mini E3 support Ender 3 V2?

HYC23, 2020-10-11:
I recently got an Ender 3 V2, but one of the stepper motor drivers on the motherboard is broken. I have an old ender 3 running SKR mini E3 board and works pretty well, so is the new SKR mini E3 V2 board support the new Ender 3 V2 printer, eg. the screen?

jernejp21, 2020-10-11:
I have just configured Marlin for Ender 3 V2. You also have to do some pin changes on ribbon cable for display, because pinout is different. I will update this post next week.

Currently I have some problems with USB connection to PC. Windows don't show anything in device manager. But printing and display works fine.

mustang5269, 2020-10-11:
I was actually looking to do the opposite. Ass the ender3 v2 screen to an ender 3 pro with skr mini e3 v2. Curious on the pin changes needed to get this to work.

HYC23, 2020-10-12:
> I have just configured Marlin for Ender 3 V2. You also have to do some pin changes on ribbon cable for display, because pinout is different. I will update this post next week.
>
> Currently I have some [...]

Would you mind showing us the pinout of the Ender 3 V2 screen and how to change the pinout?
I cannot find any info for that.
Thank you so much.

jernejp21, 2020-10-12:
You need to import this code into `pins_BTT_SKR_MINI_E3_common.h` file. I pasted this part of code below *USB connect control*. This code can be found in `pins_CREALITY_V4.h`.

```C
/**
 *        Ender 3 V2 display                         SKR Mini E3 V2.0
 *                _____                                     _____
 *            5V | 1 2 | GND                            5V | 1 2 | GND
 *   (BTN_E1) A  | 3 4 | B (BTN_E2)         (BTN_EN1) PB15 | 3 4 | PB8 (BTN_E2)
 *          BEEP | 5 6   ENT (BTN_ENC)                PB9  | 5 6   RX1
 *  (SKR_RX1) TX | 7 8 | RX (SKR_TX1)                RESET | 7 8 | TX1
 *            NC | 9 10| NC                  (BEEPER) PA15 | 9 10| PB5  (BTN_ENC)
 *                -----                                     -----
 *                EXP1                                      EXP1
 */
#if ENABLED(DWIN_CREALITY_LCD)

  // RET6 DWIN ENCODER LCD
  #define BTN_ENC                           PB5
  #define BTN_EN1                           PB15
  #define BTN_EN2                           PB8

  #ifndef BEEPER_PIN
    #define BEEPER_PIN                      PA15
    #undef SPEAKER
  #endif
#endif
```

Commented picture shows pinout of display and SKR board.

**CAUTION!!!** Connect SKR RX to display TX pin and SKR TX to display RX pin! This is the most important part (and setting the correct serial port in Marlin).

You can connect other pins as you like, just make sure you define correct pins (you can connect display A to SKR's PB9 etc.).

**Pinout for Ender 3 V2 display is printed on display's PCB. If you remove the back cover, you will see the connector marked.**

After you have copied this code, adjusted pins on display and adjusted pins for button and beeper, change also `Configuration.h` file.
- set `#define SERIAL_PORT_2 1`.
- set `#define DWIN_CREALITY_LCD` and comment out current display you are using.
- you might have to disable some functions on display. CR10 display offers more functionality than Ender 3 V2 display.

I just jumper cables to connect and test. You might find better pinout solution for button and beeper pins. TX and RX pins, you cannot change.

I hope that pinout in the comment is understandable. If it's not, please advice how to correct it to make it more readable.

Stalkeros000, 2020-10-19:
Hi all.The day before yesterday I received this board, killed two days on the configuration, was already going to sell it, when I suddenly came across this post. Thank you very much, you have given me hope. 
Please tell me in file platformio.ini you indicate default_envs = STM32F103RC_btt_512K and in configuration.h MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V2_0 right?

jernejp21, 2020-10-19:
> Please tell me in file platformio.ini you indicate default_envs = STM32F103RC_btt_512K and in configuration.h MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V2_0 right?

Yes, that is correct.

Hope you get it to work. Please comment on the results.

Stalkeros000, 2020-10-20:
> > Please tell me in file platformio.ini you indicate default_envs = STM32F103RC_btt_512K and in configuration.h MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V2_0 right?
> 
> Yes, that is correct.
> 
> Hope you get it to work. Please comment on the results.

Thx, but i cant' compile FW file, few days ago i was create new issue (before I was see that post): `https://github.com/MarlinFirmware/Configurations/issues/274`

Maybe need in configuration.h indicate CREALITY_V4?

jernejp21, 2020-10-20:
@Stalkeros000 did you look at a post above where you have to add some code to `pins_BTT_SKR_MINI_E3_common.h`?

dani190, 2020-11-01:
@jernejp21 just to confirm, you would need to place that code in the correct file and build a custom cable to wire it as defined? So basically you can't just use a regular flat ribbon cable for this to work?

jernejp21, 2020-11-01:
@dani190 yes. You have to make new cable or adapter to use with default cable, because pinout of CR10 display differs from Ender 3 V2 display.

dani190, 2020-11-01:
Ok fair enough - and this will use the stock Creality menus I guess? Since the marlin UI has not been re-written yet?

jernejp21, 2020-11-01:
I have will contact Marlin developers if they want to include this fix or not. You will need to fix HW by yourself (or if BTT or someone else starts to sell adapters). This configuration isn't plug and play like it's for BLTouch and Neopixle and sensorless homing.

Stalkeros000, 2020-11-01:
Hi all. Im not test new board yet, bcs I dont want lose PC connection. Sometimes need use some G-code commands in real-time. Only that stops me to trying made some adapter and change board. If you know how fix that problem, please tell.

dani190, 2020-11-02:
> 
> 
> I have will contact Marlin developers if they want to include this fix or not. You will need to fix HW by yourself (or if BTT or someone else starts to sell adapters). This configuration isn't plug and play like it's for BLTouch and Neopixle and sensorless homing.

Weird that this worked for you... I tried it and got a ton of errors about the encoder wheel or something. I ensured I put the above code into the file you mentioned. Were there any other changes you made?

jernejp21, 2020-11-02:
Check if everything is defined. Also check for `#define DWIN_CREALITY_LCD` in `Configuration.h`.

dani190, 2020-11-02:
@jernejp21  Ok so it seems like the stock Marlin config files are missing some stuff that must be turned on... Fixed it by comparing to the Ender 3 V2 example, enabling a few things for the encoder wheel and didn't have classic jerk enabled causing other issues. Going to try building a cable now and see how it works

MrSalzi, 2020-11-03:
after flashing my printer keeps booting. It shows progress bar and after it's full it starts again from the beginning.

dani190, 2020-11-03:
Ya I had the same behaviour... gave up for now as I know Marlin's dev's are working on fixing this soon.

jernejp21, 2020-11-03:
@MrSalzi @dani190 I have updated my comment on how to setup config file.
Did you `#define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V2_0` in Configuration.h? From @MrSalzi comment I think that wiring is wrong. I think that display pulls down REST pin so the main MCU resets. To check my theory, please use jumper cables and connect only +5V, GND, TX and RX pins. If printer boots up correctly (you can see main screen), than you have wiring problems. Check actual wiring and check how pins are set in FW.

jernejp21, 2020-11-03:
Update regrading Marlin support for this configuration.
As you can read [here](https://github.com/MarlinFirmware/Marlin/issues/19858), Marlin is not in favour of this, because it needs HW fiddling. It's not plug and play solution. In my opinion a lot of things Marlin supports are not plug and play, but this is their response.

It's like the time you needed adapter for Ender 3 (8-bit board) for using BLTouch.

dani190, 2020-11-04:
> @MrSalzi @dani190 I have updated my comment on how to setup config file.
> Did you `#define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V2_0` in Configuration.h? From @MrSalzi comment I think that wiring is [...]

yes I had it defined from before. Took a working copy of my config and just added your suggested changes
[image]

dani190, 2020-11-04:
> Update regrading Marlin support for this configuration.
> As you can read [here](https://github.com/MarlinFirmware/Marlin/issues/19858), Marlin is not in favour of this, because it needs HW fiddling. [...]

Well I don't know if that means they aren't going to support it but like you said... it needs a custom cable.

MrSalzi, 2020-11-04:
Thanks for your fast response. I've done all that and tried to disconnect all the other pins but these didn't fix it :(

MrSalzi, 2020-11-04:
Is this correct?

[image]

[image]

MrSalzi, 2020-11-06:
If I comment out #define EEPROM_SETTINGS then it boots to main menu. Button works but rotation doesn't, but anyway my board is alive.

Westward12, 2020-11-18:
> If I comment out #define EEPROM_SETTINGS then it boots to main menu. Button works but rotation doesn't, but anyway my board is alive. I guess this isn't the final solutions?

I am at the same step. I can get it to post by commenting out EEPROM_SETTINGS. With it enabled it just boot loops.

Madtom2925, 2020-12-14:
Ok, so I got it to work with eeprom, and with almost everything working including z offset and a bltouch. There is a flaw in the code for saving the language setting to the eeprom that makes it only work with the eeprom on the creality v4 board. With these lines commented out marlin starts properly however as a default it will always start in English. I'll do a full write up on how to make it work later when I get home.

jottr, 2020-12-14:
@bigtreetech do you plan to officially support the Ender3v2?

thomasparlette, 2020-12-16:
I am getting the following error when compiling.
[image]

Attached is also my configurations.
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5699923/Configuration.zip)

thomasparlette, 2020-12-16:
> I am getting the following error when compiling.
> [image]
> 
> Attached is also my configurations.
> [Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5699923/Configuration.zip)

Figured out the issue in Configuration_adv.h calling out the serial connections to each 2209 driver. It compiled now the new board should be in tomorrow and I can install and try out.

Madtom2925, 2020-12-19:
Hi all, Sorry for the delay in my writeup. the reason for the endless loading bar loop in the start-up cycle of the ender 3 v2 screen is due to two lines of code that are specific to the BL24CXX EEPROM and its library. I'm not sure on how to fix this so it works with all EEPROMS however I found disabling the lines of code causes the LCD to default to English. 

The Lines of code are in Marlin\src\lcd\dwin\e3v2\dwin.cpp

You will need to // line 218 and Line 227  like so:

`void HMI_SetLanguage() {
  #if ENABLED(EEPROM_SETTINGS)
    //BL24CXX::read(DWIN_LANGUAGE_EEPROM_ADDRESS, (uint8_t*)&HMI_flag.language, sizeof(HMI_flag.language));
  #endif
  HMI_SetLanguageCache();
}

void HMI_ToggleLanguage() {
  HMI_flag.language = HMI_IsChinese() ? DWIN_ENGLISH : DWIN_CHINESE;
  HMI_SetLanguageCache();
  #if ENABLED(EEPROM_SETTINGS)
    //BL24CXX::write(DWIN_LANGUAGE_EEPROM_ADDRESS, (uint8_t*)&HMI_flag.language, sizeof(HMI_flag.language));
  #endif
}`

Also if you are getting a definition error for ENCODER_5X_STEPS_PER_SEC you will need to add a line in Marlin\Configuration_adv.h.

By default it only appears to have 10x and 100x defined , I added a line like this to resolve the error.

`#if ENABLED(ENCODER_RATE_MULTIPLIER)
  #define ENCODER_5X_STEPS_PER_SEC    15  // (steps/s) Encoder rate for 5x speed
  #define ENCODER_10X_STEPS_PER_SEC   30  // (steps/s) Encoder rate for 10x speed
  #define ENCODER_100X_STEPS_PER_SEC  80  // (steps/s) Encoder rate for 100x speed
#endif`

I also got an error when trying to save to the dedicated eeprom on my skr e3 2.0. I had to activate the flash eeprom for it to work, if someone else can try this and see if it is the same for them that would be great.

thomasparlette, 2020-12-20:
Here is an edit to the code so that it will only activate this code if the BOARD_CREALITY_V4 is defined

` 
void HMI_SetLanguage() {
  #if ENABLED(EEPROM_SETTINGS)
	  #if (MOTHERBOARD == BOARD_CREALITY_V4)
		  BL24CXX::read(DWIN_LANGUAGE_EEPROM_ADDRESS, (uint8_t*)&HMI_flag.language, sizeof(HMI_flag.language));
	  #endif
  #endif
  HMI_SetLanguageCache();
}

void HMI_ToggleLanguage() {
  HMI_flag.language = HMI_IsChinese() ? DWIN_ENGLISH : DWIN_CHINESE;
  HMI_SetLanguageCache();
  #if ENABLED(EEPROM_SETTINGS)
    #if (MOTHERBOARD == BOARD_CREALITY_V4)
      BL24CXX::write(DWIN_LANGUAGE_EEPROM_ADDRESS, (uint8_t*)&HMI_flag.language, sizeof(HMI_flag.language));
    #endif
  #endif
} 
`

Madtom2925, 2020-12-20:
That works for automatically disabling the code if its not the right motherboard, I was more thinking of a way to allow it to save using marlins eeprom definitions so it would work with any board rather then just being disabled. That way people who want the other languages on boot can use it.

davthomaspilot, 2020-12-27:
> Update regrading Marlin support for this configuration.
> As you can read [here](https://github.com/MarlinFirmware/Marlin/issues/19858), Marlin is not in favour of this, because it needs HW fiddling. [...]

jernejp21 ,

You haven't commented lately.  It sounded like you were successful in getting the BT mini E3 V2 to work in an Ender 3 V2.  Wiring between the display and the board needed changes, but with the required firmware changes you documented, everything worked fine for you?  But, I see issues others have had since you posted.  Did you not have these issues?  

My Ender 3 V2 board died after less than a week of use, so I ordered the mine E3 V2 as a replacement.  Maybe I'll just leave the display unplugged and use an RPI for octoprint.  Any reason that won't work?  Must a display be plugged into the E3 V2 board for things to work properly using Octoprint?

I'm quite comfortable with doing a custom cable, but I figure I should do the upgrade in smaller steps, if there are numerous issues to resolve.

jernejp21, 2020-12-28:
I didn't have any issues. I just changed ribbon cable from board to display and edited Marlin as described in post above. BL touch is working, sensorless homing is working, display is working.

davthomaspilot, 2020-12-28:
Thanks!  Your posts seem to make it straight forward.  I'll follow them tomorrow when my BT E3 V2 arrives.

davthomaspilot, 2020-12-30:
Thanks for the reply!
Got my BTT board for my E3 V2 and ready to and  went back and read your post.  I noticed you had issues with USB.  Did you resolve that?
Dave Thomas

jernejp21, 2021-01-01:
USB issues will remain, because of faulty SKR design. SCLK pin is connected to USB D+ pin via MOSFET. This means that you cannot use USB when you are debugging the board.

davthomaspilot, 2021-01-01:
So, it just impacts debugging?  No issues with printing from a PC like Octoprint?

davthomaspilot, 2021-01-02:
jernejp21 

Your instructions said:

> set #define SERIAL_PORT_2 1.

But that means SERIAL_PORT must be changed to something other than 1.  Else, I get these errors:

#error "SERIAL_PORT_2 cannot be the same as SERIAL_PORT

Did you mean 

set #define SERIAL_PORT_2 -1

If I do this, everything builds ok.   

Or, should I #define SERIAL_PORT to something besides 1

jernejp21, 2021-01-02:
SERIAL_PORT -1
SERIAL_PORT_2 1

davthomaspilot, 2021-01-02:
Thanks again!

I found a set up guide here:

[https://www.makenprint.uk/3d-printing/3d-printing-guides/skr-mini-e3-v2-configuration-h-marlin-2-setup-part-2/](url)

It says to use 

SERIAL_PORT 2
SERAIL_PORT_2 -1

The comments in the code say:

`/**
SERIAL_PORT
 * Select the serial port on the board to use for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port -1 is the USB emulated serial port, if available.
 * Note: The first serial port (-1 or 0) will always be used by the Arduino bootloader.
 *
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */

SERIAL_PORT 2
/**
 * Select a secondary serial port on the board to use for communication with the host.
 * Currently Ethernet (-2) is only supported on Teensy 4.1 boards.
 * :[-2, -1, 0, 1, 2, 3, 4, 5, 6, 7]
 */

'
The comments suggest both SERIAL_PORT and SERIAL_PORT_2 define what to use "for communication with the host"I don't know what "secondary" serial port means.   Maybe there are two serial ports on the SKR card, one is used for the USB connection and there's a header for debugging over a serial interface?  Sorry for my ignorance.

It also says it's "critical" to change the driver types to TMC2209 (versus TMC2208_STANDALONE).

```
#define X_DRIVER_TYPE  TMC2209
#define Y_DRIVER_TYPE  TMC2209
#define Z_DRIVER_TYPE  TMC2209
//#define X2_DRIVER_TYPE TMC2209
//#define Y2_DRIVER_TYPE A4988
//#define Z2_DRIVER_TYPE A4988
//#define Z3_DRIVER_TYPE A4988
//#define Z4_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2209
//#define E1_DRIVER_TYPE A4988
//#define E2_DRIVER_TYPE A4988
//#define E3_DRIVER_TYPE A4988
//#define E4_DRIVER_TYPE A4988
//#define E5_DRIVER_TYPE A4988
//#define E6_DRIVER_TYPE A4988
//#define E7_DRIVER_TYPE A4988
```

Let me know if this is the wrong place to ask these questions.

Madtom2925, 2021-01-02:
I'm fairly certain that guide is for an ender 3 v1, not a v2. You need serial_port_2 defined as 1 so that it uses the correct pins in the exp header for the display as it uses serial_port_2 for the display.  The rest with the tmc drivers is correct, you also need to use classic jerk and make the changes to the code as listed above.

davthomaspilot, 2021-01-02:
Thanks! 

 It would be nice if there were symbolic constants or an enumeration for the values of SERIAL_PORT and SERIAL_PORT_2.  Like 
`#define USB_EMULATED -1`
and, based on Madtom2925's reply
`#define DISPLAY_SERIAL 1`

or maybe there are different valid values for SERIAL_PORT versus SERIAL_PORT_2?  What would "2" indicate if used for SERIAL_PORT_2?  Are the valid values for each constant documented somewhere in the code (or elsewhere)?

I'll give 

```
SERIAL_PORT -1
SERIAL_PORT_2 
```1
a try with the rewired display.

CLASSIC_JERK is defined in the sample configuration file jerniejp suggested to use as the starting point, but the DRIVER_TYPES are defined as TMC2208_STANDALONE and apparently need to be changed.

I think I want to understand the Marlin code better.  Are there any resources besides just start reading the code?

Thanks again!

jernejp21, 2021-01-02:
STM32F103 has 4 available USART pin pairs (TX, RX pin pairs). -1 means to use USB UART (D+, D- pins). 1 means TX1, RX1, 2 is TX2, RX2 and so on.
See reference manual which pin is which USART. 

There is no documentation about how Marlin is structured or any program flow. You just have to start reading the code.

davthomaspilot, 2021-01-02:
Thank you jernejp21!  That helps a lot.

Jd1Ent, 2021-01-15:
Hello,
I have been following along with this thread. I recently installed an SKR mini E3 v2.0 on my Ender 3 V2. With your help and Youtube I have gotten to the point where everything is working except, when I try to print something, the printer will auto home, the progress bar for the print will go to %100 very quickly but the printer never moves after that. Also when try to bed level with the BL Touch, it probes once and then stops moving. 

 Any suggestions are greatly appreciated.

Ogrenero, 2021-01-15:
hi everyone, I'm new to this 3d world and unfortunately by making some mistakes I broke the motherboard of my ender 3 v2, today I tried to replace it with the skr mini e3 v2.0 with the stock lcd, I state that I am not practical of code, so I was looking for someone who could help me with the first configuration. one more thing, is it normal for me to beep continuously?
thank you so much

Madtom2925, 2021-01-15:
> I have been following along with this thread. I recently >installed an SKR mini E3 v2.0 on my Ender 3 V2. With >your help and Youtube I have gotten to the point where >everything is working except, [...]

We would need to see your config and advconfig files to see what's going on.

Ogrenero, 2021-01-15:
> We would need to see your config and advconfig files to see what's going on.

no the problem is that I haven't loaded any firmware yet because I don't know which one to put, but something else "do I have to follow the lcd display output configuration described above?"
sorry for my English

Madtom2925, 2021-01-15:
That was in reply to the problem quoted, not you. For this to work you have to modify the cable as listed above. Otherwise it will just beep. You will also need modified firmware, do you have a bltouch?

jernejp21, 2021-01-15:
Sorry to everyone it took some time. I have updated my original post with all the settings I have modified from Ender 3 V2 config file. 

I see a lot of people are asking same questions over and over:
- you have to create new ribbon cable, because pinout if different. DWIN display has different layout than CR10 display.
- if your display is beeping, pinout is incorrect.
- if you can't see image on display, pinout or FW is incorrect
- if FW doesn't compile read all error messages. You might have multiple displays defined or you don't have the correct board selected in platformio.ini file.

Madtom2925, 2021-01-15:
Jernejp21 do you have eeprom enabled?

jernejp21, 2021-01-15:
You need EEPROM to store settings and to use M500 command.

Ogrenero, 2021-01-15:
> Era in risposta al problema citato, non tu. Affinché questo funzioni è necessario modificare il cavo come elencato sopra. Altrimenti emetterà solo un segnale acustico. Avrai anche bisogno di un firmware modificato, hai un bltouch?

yep i have bltouch

Madtom2925, 2021-01-15:
Jernejp21 did you have to make mods to the dwin file or did it work straight away for you?, some people seem to get a boot loop and others don't so maybe its version related

Jd1Ent, 2021-01-16:
> We would need to see your config and advconfig files to see what's going on.

I uploaded the config files to my profile in   "skr-e3-v2.0-on-ender-3-v2"

Madtom2925, 2021-01-16:
I don't see anything obvious in the config,  how is your bltouch wired?, is it plugged in to the z min header or the bltouch header?

Jd1Ent, 2021-01-16:
I have it on the one labeled z probe.

Jd1Ent, 2021-01-16:
> I don't see anything obvious in the config,  how is your bltouch wired?, is it plugged in to the z min header or the bltouch header?

I ended up disabling bl touch in firmware and now it will print, so I will have to dig into the bl touch settings to see what is going on.

[106 more replies]