bigtreetech/BIGTREETECH-SKR-mini-E3 #426: [BUG] Cannot set my z-offset

darthdively, 2020-10-12:
hey folks, i could greatly use some help. first off let me tell you my skill level. I'm one of those that can learn things if given the right tools and pointed in a good direction. can figure stuff out too. I'm not a programmer. I can download files and load firmware. Now that's out of the way, I'm having problems with the SKR Mini i just got.  SKR Mini e3 v2 with BL touch V1 on ender 3.
current goal: Setting Z Offset
i uploaded the latest firmware of marlin 2.0.6 from github. the file i used is the firmware for BL touch with z homing. I had to use the connections with the white/black z stop and the 3 yellow/red/blue on the z-probe.  using the new interface from the firmware on the ender 3, i'm trying to move the z axis but it won't let me go into the negative values to bring it closer to the bed. i'm about to pull out my hair. this is so frustrating.  can anyone please help?
Thank you

Darkwho58, 2020-10-14:
Hi, I have the same problem as you! Impossible to make the Z value negative ... While waiting for a solution, I kept the Z sensor and put the BLT aside! I tried to compile a Firmware myself but without success for me.

if anyone can help us?

abacabie, 2020-10-14:
If you can connect to your PC and use pronterface or use octoprint you can send M211 S0 to disable end-stops, this will let you move Z axis into negative, use M211 S1 to enable them again.

Darkwho58, 2020-10-14:
Thank you for the info.

except that I have not yet octoprint (I am still a novice in it) ...
and to tell the truth I do not know exactly what its use, hence the use of Firmware already close.

I took a skr mini e3 v2 especially for the silence of the engine, the possibility of a double Z axis and the possibility of a BLT.

madeinholland, 2020-10-14:
I have spent the holl day trying to make it work and now I am reading this .....

madeinholland, 2020-10-15:
I have already tried it with the marline folder. but I set the BL-touch to my z-as and the other pin in the original place.
but it still happens to me.  

I have no issues if I use the original firmware, but I wanna set the BL-touch and the runout filament with auto home-ing.

darthdively, 2020-10-27:
has anyone got any updates with this yet?

abacabie, 2020-10-28:
If you can connect to a pc or laptop you can use Pronterface to disable end-stops, i believe that's the only way of doing it.

madeinholland, 2020-10-29:
> If you can connect to a pc or laptop you can use Pronterface to disable end-stops, i believe that's the only way of doing it.

so I need a Raspberry pi to fix the code

abacabie, 2020-10-29:
You can use a usb lead between your printer and a pc or laptop and pronterface to send commands, have a look at Teaching tech on you-tube, he has a mountain of guides that are easy to follow and covers this, its sometimes easier if you can actually see it.

darcNL, 2020-11-10:
> 
> 
> If you can connect to your PC and use pronterface or use octoprint you can send M211 S0 to disable end-stops, this will let you move Z axis into negative, use M211 S1 to enable them again.

abacabie!!! Omg thank you. This is exactly what I've been looking for. This has allowed me to adjust my z-axis into the negative, to find my sweet spot and adjust my z-probe offset. This only took me 24hrs to do. Omg you don't even know. Thank you!

darthdively, 2020-11-13:
Okay I just got this working now....but.... I bought the TFT35.  So with the TFT35 installed I got it to work--
1)installed TFT35 and made sure the BL touch connections were placed where needed. the black and white cable connected to the z-stop in the board and the other 3 colors plugged in the z-probe spot that's shown in online tutorials
2)updated firmware with the latest 2.0.7.1? or so whichever is latest one that was available within this last week. *use the .bin file that uses z-probe homing
3)turned off the printer and turned it back on
4)homed the printer
5)tried to move z-axis to negative value.  IT DID NOT WORK.  Brought it to zero(still space between the printer and the nozzle).
      -KEEP THE HOT END/NOZZLE WHERE IT IS and turn off the unit.
6) Turn on printer, DO NOT HOME
7) attempt to move the z-axis to negative value
All of this worked for me.  If it doesn't work for you, I don't know what you can do . but it seems to work well when paired with the TFT35.  I have not tried this process with the original display.

After moving z-axis to negative value that brings you close to bed, go into settings and set it as your z-offset.  doing test prints and use the baby stepping feature, do some simple arithmetic to find your perfect z-offset value.

I noticed that this firmware is more friendly with the TFT35 rather than the original display screen.  the 35 bucks on amazon is really worth it to pair with this board. i should have done it in the first place.  It allows you to change certain features or settings all from the TFT35. For example setting the probe X and Y offsets with this firmware and board you can do easily with TFT35. If you tried to change it in the original display screen, it will give you an error.

I hope this helps somebody. If someone can try this process with the original display screen and let us know.  I may try to do it myself but I've been wanting to get printing for a long while and I don't want to mess anything up now lol.

Good luck, please let us know your results.