bigtreetech/BIGTREETECH-SKR-mini-E3 #427: SKR Mini E3 v2.0 and BLtouch

FrankAndrade1971, 2020-10-14:
I installed a SKR Mini E3 v2.0 on my ender 3 pro and hooked up a bltouch. Went to flash the firmware from bigtreetech from github  and got the follow errors on startup:
- Failed to enable bed leveling
- Bed Leveling Off
- Fade Height OFF

Did i do something wrong or is it just the firmware is not correct?

Slimgig, 2020-10-14:
No it does it to mine every time I boot up it's because the screen is
active before the board is so it give you that error  does it say ABL
active after them errors like mine does and you have to go in and set the
fade height that's y it gives you that error too but mainly  the error are
from the screen looking for them commands but the board asnt booted yet
that's all it is mine does it even time I turn on my machine on

Slimgig, 2020-10-14:
It will do that evertime you boot but everything thing will work just make sure you go to abl leveling and do the z-offset and set that then do a bed leveling

cobramike, 2020-10-15:
Nope not working. With the BLTouch v3.1. Fra

> I installed a SKR Mini E3 v2.0 on my ender 3 pro and hooked up a bltouch. Went to flash the firmware from bigtreetech from github and got the follow errors on startup:
>
> * Failed to enable bed [...]

Nope not working. With the BLTouch v3.1. i have the same problem. So there is somthing wrong with the firmware.

mustang5269, 2020-10-15:
I believe you have to do a bed level first for that message to go away.

Fredom-soldier-cz, 2020-10-17:
> I believe you have to do a bed level first for that message to go away.

Yes. You have to start leveling manually from menu qnd save. Then this messages disappear.

JPMATIZ, 2020-10-28:
I Installed an BTT SKR MINI E3 V2.0 and the BL on a Ender 3 Pro but when I try to do a Z Homing the Z axis is far apart from the bed and the BLT activate 3 times , and then appear a M999 BLtouch error . So I check the wiring and its good . I Uploaded the "firmware-bltouch-for-z-homing.bin"  and the problem still appear . So I Configured the Marlin again and uploaded but the error appear again . Any ideas what could be the  problem ?

dahose, 2020-10-30:
BTT firmware on github did not work well for me originally, but that was with version 2.0.6.  I installed the bltouch for z homing 2.0.7 firmware, and it worked fine.  Be sure you have good connection on z-axis limit port, and be sure power connection is correctly oriented.  When printer boots, go to BLT settings and run test.  The BLT should cycle a few times and finish OK.  If the test runs fine, then set your offset [following this great video by 3DMN](https://youtu.be/y_1Kg45APko).

Jose

Florecs71, 2020-12-02:
> I Installed an BTT SKR MINI E3 V2.0 and the BL on a Ender 3 Pro but when I try to do a Z Homing the Z axis is far apart from the bed and the BLT activate 3 times , and then appear a M999 BLtouch [...]

He does the same for me. How did you solve the problem in the end?

Florecs71, 2020-12-13:
> I Installed an BTT SKR MINI E3 V2.0 and the BL on a Ender 3 Pro but when I try to do a Z Homing the Z axis is far apart from the bed and the BLT activate 3 times , and then appear a M999 BLtouch [...]

Did you solve the problem? I have the same mistake and I don't know what to do. Please help me if you can.

Angus442, 2020-12-28:
Hello, i managed to fix this problem by 1. when the 3d printer is OFF pull down the BL touch pin  2. turn on BL touch it the pin should go up down up down etc 3. Do ABL leveling 4. BL touch press deploy then stow 5. then do ABL leveling BL touch "start" 6.  wait till all 25 homing things are done 7. hit save to epprom 8. reboot 3d printer 9.Done error FIXED and BL touch errors FIXED 

This soultion worked for me and yes im on the latest Bl touch for z homing bin with skr mini e3 v2 on a ender 3 pro so i did have the exact error / problem as that user above me but i managed to fix it with what I  did in the steps I said above. 

- Angus 
Happy printing.
  :)

Florecs71, 2020-12-28:
Thanks Angus, I will try but only at the beginning of next year. I'm currently busy with other issues. I keep you informed and thank you for your attention. I wish you a happy new year!

pablo123b, 2021-01-11:
hi guys newbie alert, so be patient lol. i have the ender 3pro skr mini E3 vr2.o, with the tft353 screen and now am adding the bltouch.
The problem i am having is flashing the firmware yes tried all the file and change the name to firmware.bin  still no joy anybody that can hell it would be very much appreciated thx and stay safe

YBeyin, 2021-01-12:
> hi guys newbie alert, so be patient lol. i have the ender 3pro skr mini E3 vr2.o, with the tft353 screen and now am adding the bltouch.
> The problem i am having is flashing the firmware yes tried all [...]

Are you using the sd card slot whic one is on the mainboard? You must to use that for mainboard firmware update.

pablo123b, 2021-01-24:
> Are you using the sd card slot whic one is on the mainboard? You must to use that for mainboard firmware update.

no i hadn/t i was using a extension and when you mentioned that it worked so thank , sorry for delay in reply

pablo123b, 2021-01-25:
> Hello, i managed to fix this problem by 1. when the 3d printer is OFF pull down the BL touch pin 2. turn on BL touch it the pin should go up down up down etc 3. Do ABL leveling 4. BL touch press [...]

hi angus, we have the same set up, but mine now throws up ml999 which is a pain as first time it worked correctly!! ggrrr

lu9dqk, 2021-02-19:
Hola: me pasaba exactamente lo mismo que a Ustedes, mi impresora es Ender 3 con auto nivel Touch-Mi fr.
tarjeta BIGTREETECH-SKR-mini-E3 V2.0 + Pantalla Pantalla táctil LCD TFT35 3.0 y lo solucione presionando la rueda 4 segundos y pasando a Marlin en la pantalla y ahí fui al menú de Touch-Mi  le realice los ajustes con los controles y luego de todo el proceso pase nuevamente a la pantalla táctil  y solucionado. Espero puedan solucionarlo, yo de esa manera estoy muy contento que funciona. Saludos

mark-hahn, 2021-07-07:
> Hello, i managed to fix this problem by 1. when the 3d printer is OFF pull down the BL touch pin 2. turn on BL touch it the pin should go up down up down etc 3. Do ABL leveling 4. BL touch press [...]

I tried to follow these instructions but I cannot ABL in step 3 because the z-axis isn't homed.  I can't home the z-axis because of the error.  Any ideas?

SeanDickson, 2021-12-14:
Did you get this fixed?

jakedoesgithub, 2022-08-03:
Hi it’s not you it’s the recompiled firmware it’s incorrect here are the older btt version that work https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/d5fc619c2a4aadc889c47680af351310f521e850/firmware/V2.0