bigtreetech/BIGTREETECH-SKR-mini-E3 #428: [BUG] (SKR mini E3 v1.2 no flash with SD )

Alexkey2200, 2020-10-16:
Hi, I´ve  a SKR mini E3 v1.2 whit BlTouch. But ....

I have uploaded the marlin firmware several times modifying the autobed leveling, bootscreen and grid parameters to my liking. But this last time I needed to reverse the rotation of the extruder motor but when uploading it to SD the SKR does not take the new firmware. And tried everything in the forums, from formatting the SD, updating VStudio and the Platformio Core, restoring the default settings in the EEPROM, disconnecting the BLtouch. But nothing works for me, it should be noted that the sd reader of the plate works since I always print from there.

Does someone know how to solve this problem?

Switchcrafter, 2020-10-17:
can you expand on "SKR does not take the new firmware"?  

After booting off the SD card containing the new firmware.BIN,  does the file extension change to firmware.CUR after it boots?

Alexkey2200, 2020-10-18:
Hi, Switchcrafter.

       Yes te firmware change to te firmware.CUR after it boots. I managed to find the problem and it turns out that I had corrupted marlin sources for some reason (I have them in OneDrive and I think the modifications on the disk were not updated), I downloaded the marlin 2.07 file and it went up perfectly without problems.

closed 2020-10-18 by Alexkey2200