bigtreetech/BIGTREETECH-SKR-mini-E3 #430: [BUG] Part cooling fan problem on firmware 2.0.7.2

altadeos, 2020-10-17:
Hello,

I have a problem with the part cooling fan. It won't start or update the speed until one enters and exits various menus.

If I pushes the knob to enter the menu during the print, the fan suddenly starts at the very exact moment I push the knob. Same goes when the fan speed changes, Speed-changes doesn't "go through" until the knob/menu is triggered.

I tried with various slicers and the problem is the same.

Could you please correct this bug ?

Best regards,

bnutz2k, 2020-10-18:
I can confirm this or similar behavior when I tried to load the latest firmware-bltouch.bin. The blower fan manual controls do not work (or at least they don't always respond). It would only turn on for me if I heated the bed. Bad news.

Also had an issue with z movement not responding to manual controls. 

Rolled back and those issues went away.

Ender 3 v1
SKR Mini E3 V2

Thegs68, 2020-10-20:
Hi
I have the exact same problem as @altadeos with the part fan only turning on when you push the Menu knob. This has only happened since I upgraded to 2.0.7.2 with my SKR Mini E3 v2 board. I compiled my own firmware file but see no issues with missed options. I also tried the firmware-bltouch-for-z-homing.bin same problem.

Please Help!

Stay Safe, Stay Healthy

Ender 3 Pro
BigTreeTech SKR Mini E3 v2.0
BL-Touch Smart v3.1

kjbuente, 2020-10-20:
Can also confirm. Sending M106 does not kick fan on at all, while the display does show 100%. The fan does blip for a second as the Buzzer beeps while navigating through the menu. I'm able to reactivate mine if I go into Tune, tweak the fan setting and then back out into the main menu. Then it only seems to work for a few minutes.

Ender 3 Pro
SKR Mini E3 V2
BL-Touch v3

LearnIIBurn, 2020-10-20:
Having somewhat similar issues as well. Fan does not turn on or respond to manual percentage changes or turn on when print starts or any heating is turned on. Basically the part cooling fan is inactive completely. I've been reading that the fan mosfets can be pretty bad on these boards but since this seems to be an issue that came up after Marlin update 2.0.7.2 I think it's probably firmware related.

altadeos, 2020-10-20:
After downgrading to Marlin 2.0.X bugfix (May version) the problem disappear.

abacabie, 2020-10-20:
Running the most current version of marlin bug fix with the skr mini v2 (ender 3 pro) and i`m not seeing this issue, for me its not present in the most recent bugfix.

Thegs68, 2020-10-21:
I can confirm @abacabie that flashing the latest Big Fix does actually fix the parts fan problem of not turning on at a certain height or layer. I did this last night and everything seems to work well.

Ender 3 Pro
BigTreeTech SKR Mini E3 v2.0
BL-Touch Smart v3.1
Octo Print 0.17.0

Blacksource001, 2020-10-22:
Can someone send me the good version of firmware-bltouch-for-z-homing.bin
Thanks!

Tigui91racing, 2020-10-23:
Searching for a good version with fan correction for my ender 3 with BLtouch z homing ! thanks !

Thegs68, 2020-10-23:
Here is a link to my working 2.0.7.2 firmware-bltouch-for-z-homing.bin file that I compiled myself. Works for fan control as well. Hope it works for you.
[https://drive.google.com/file/d/1Q52uWfnoIGPzZicjXR8FDcFilRWgXKJg/view?usp=sharing](url)

jorgeasl, 2020-10-23:
Same issue here, part cooling fan not working after upgrading my SKR Mini E3 V2 to the latest firmware-bltouch.bin available here. 
Downgraded to the May release, fan works fine.

Dees anyone have a working version of firmware-bltouch.bin? 
thanks

dmmatos, 2020-10-23:
Do you think this will work for Ender 3 (not pro). I finally got my SKR v2.0 installed, got BLTouch tuned well and my parts fan doesn't turn on anymore.

> 
> 
> Here is a link to my working 2.0.7.2 firmware-bltouch-for-z-homing.bin file that I compiled myself. Works for fan control as well. Hope it works for you.
> 
> https://drive.google.com/file/d/1Q52uWfnoIGPzZicjXR8FDcFilRWgXKJg/view?usp=drivesdk
> 
> Ender 3 Pro
> BigTreeTech SKR Mini E3 v2.0
> BL-Touch Smart v3.1
> OctoPrint 0.17.0

Thegs68, 2020-10-23:
Hi @dmmatos
This should work for the Ender 3. Just check your Probe Offset and Z Offset are set correctly for your machine in the settings configuration, probe off set menu. I always copy mine before an upgrade to give me a starting point. All other settings should be good.

dmmatos, 2020-10-23:
Thank you very much.

I'll give it a try before I try to develop my own firmware.

timqui, 2020-10-24:
I see same problem here. New firmware = new bug again... :-/

BTT SKR mini E3 v2, Ender 3 (no BL touch or anything)

mifraburneo, 2020-10-25:
Same problem here. Also the fan won't start unless the setting is around 80% or more. When it's set to i.e. 33% (for first layers of print) it won't start at all. When I click on the menu wheel I can notice a little wiggle on the fan (like it's trying to start but it won't).
I have a brand new BTT SKR mini E3 v3.0 with BL Touch as Z home.
Would anyone suggest to use another version? Which one? I'm new to this world of alternate motherboards.
Thanks!

mifraburneo, 2020-10-25:
> Can someone send me the good version of firmware-bltouch-for-z-homing.bin
> Thanks!

If you manage to get it, please reefer it to me too!

le1parrain, 2020-10-25:
Same here. It's really irritating as I have to keep watching if the fan is working! How do I get the old firmware as only the new ones are available here?

mifraburneo, 2020-10-25:
> Here is a link to my working 2.0.7.2 firmware-bltouch-for-z-homing.bin file that I compiled myself. Works for fan control as well. Hope it works for you. [...]

This one works like a charm!!! Thank you so much!

Do you perhaps could provide us the pre-compiled project? It'd be highly appreciated! :)

le1parrain, 2020-10-25:
> This one works like a charm!!! Thank you so much!
>
> Do you perhaps could provide us the pre-compiled project? It'd be highly appreciated! :)

Thank you. I'm not using BLTouch as z homing. Do you know what settings to check in the editor?

mifraburneo, 2020-10-25:
> Thank you. I'm not using BLTouch as z homing. Do you know what settings to check in the editor?

Actually I don't know too much about the compiling process, but I was following this video as for now: https://www.youtube.com/watch?v=_IafavTl1fA

I think it might help you. 
You need to get the Marlin "Bugfix" version https://marlinfw.org/meta/download/
Then replace the Configuration.h and Configuration_adv.h files with your machine specifics here: https://github.com/MarlinFirmware/Configurations/tree/bugfix-2.0.x/config/examples
(search for your brand, model and specific board)

Then you follow along with the video, sort of.

dmmatos, 2020-10-25:
In having trouble with Z axis homing. I'll get it zeroed in and it will
print just give and then the next print it starts about a centimeter up.

I have my offsets all figure out.  What procedure do you use when initially
setting the home offsets?

I may just wipe the eeoprom and try again. I have all the numbers (and this
firmware).

Yesterday I couldn't even get the BLtouch to run a bed level because it was
so far away from the bed.

Thegs68, 2020-10-25:
> Thank you. I'm not using BLTouch as z homing. Do you know what settings to check in the editor?

For the z Homing you can turn off this section:
/**
 * Enable this option for a probe connected to the Z-MIN pin.
 * The probe replaces the Z-MIN endstop and is used for Z homing.
 * (Automatically enables USE_PROBE_FOR_Z_HOMING.)
 */
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN

I have never tried to use any other setup, so I don't know if there is another setting to check or if the default is the mechanical Z-endstop switch.

Thegs68, 2020-10-25:
> In having trouble with Z axis homing. I'll get it zeroed in and it will print just give and then the next print it starts about a centimeter up. I have my offsets all figure out. What procedure do [...]

Not sure why that is happening. I use Cura as a slicer and have set my G-Start Code to look like this:

G92 E0 ; Reset Extruder
G28 ; Home all axes

G1 Z10 F200 ; Move Z axis to 10mm above bed

M420 S1 ; retrieve store Mesh Data

G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

If you use OctoPrint you can also edit the "Before print job starts" setting to
G91 ; Switch to relative coordinates
G1 Z10 F200 ; Set Z axis to 10 mm above current position
G90 ; Switch to absolute coordinates

That's my settings. I'm sure that the the code for OctoPrint will also work in the Cura settings. but I always suggest using G28 to Home all axis before any Bed Level.

mifraburneo, 2020-10-26:
> Here is a link to my working 2.0.7.2 firmware-bltouch-for-z-homing.bin file that I compiled myself. Works for fan control as well. Hope it works for you. [...]

Hey Thegs68, would you mind providing your pre-compiled files of this firmware? I'd really appreciate it!

Thegs68, 2020-10-26:
Hi @mifraburneo 
Here are the files I used for compilation:
Main Files that should be put in your root directory: https://drive.google.com/file/d/1vnUbbxUG7wVCUcI7x_KmMyoVCYlMGmCT/view?usp=sharing

You will need to pick the, env:STM32F103RC_btt_512K from the Platformio list.

If needed
Configuration files for setting up your environment, (which board you will be using, pick ie: BigTreeTech SKR Mini E3 2.0 from Configurations-bugfix-2.0.x/config/examples/Creality/Ender-3 or which ever printer you have):
https://drive.google.com/file/d/1fkTU0xy58dSf_fOkJGVFc2h-2DXgNHR_/view?usp=sharing

mifraburneo, 2020-10-26:
> Hi @mifraburneo
> Here are the files I used for compilation:
> Main Files that should be put in your root directory: https://drive.google.com/file/d/1vnUbbxUG7wVCUcI7x_KmMyoVCYlMGmCT/view?usp=sharing [...]

Thank you SO much. As I mentioned earlier, I'm new to the firmware building process and since your firmware is working great for my Ender 3 Pro, I'd like to use yours as a base. 

If it's no trouble for you, would you mind mentioning which lines did you edit from the original configuration.h?

Thanks! Hopefully I won't bother you anymore with this questions :D

dmmatos, 2020-10-26:
Once I updated all my offsets, I am getting good prints. I appreciate it!

My parts fan (FAN0) still only comes on at 100% - it doesn't come on at anything less than that.

Any suggestions or is this a BTT bad MOSFET or something?

Ender 3
BigTreeTech SKR Mini E3 v2.0
BL-Touch Smart v3.1

Thegs68, 2020-10-26:
@dmmatos 
I believe that there is still a bug that is stopping the variability of the fan, I noticed the same thing . I am going to try the latest Bug Fix and see if that works, As for the configuration files here they are with the lines that I changed commented with "Thegs68" in them. So search for that and you be good to go.

https://drive.google.com/file/d/1CjSWc7ebi4BwbRJe-PvLfkgrKVk0whrn/view?usp=sharing

dmmatos, 2020-10-26:
Big Thanks Thegs68!

Hopefully the fan thing will get worked out, my first Benchy looks pretty
good. Will look through the config files and see if I need to adjust
anything else!

I truly appreciate it!

dmmatos, 2020-10-26:
Yours worked well!

I tried to update the config files and recompile to match my setup (Ender
3, BTT board, BLTouch, Microswiss Direct Drive, Petsfang mount) - basically
just changing offsets, feed rates, etc. and when

I went to compile it, it through an error - "You are using an old
Configuration file, update before building Marlin."

Suggestions? or just transfer over the changes to the standard BTT configs
on GITHub?

Again, thanks for all your help.

dmmatos, 2020-10-26:
I noticed CONFIGURATION_H_VERSION 020006  on yours and 020008 on the most
recent one.... can it just be changed?

Thegs68, 2020-10-26:
Hi
Thanks. I’m glad they worked for you. As for the version number it can’t be changed. I just went through the last eat files and there is changes and added features. So it has to be redone each time. 

Greg
Stay Safe, Stay Healthy

Thegs68, 2020-10-26:
I didn’t think there would be a prerequisite to adding a new hotend. LOL. That will make it fun. 

Greg
Stay Safe, Stay Healthy

menozcm, 2020-11-12:
Thegs68  i used your firmware.bin but the bed size seems to be set to x=220 y=235

Cathprotech, 2020-11-12:
I updated my firmware last night (9th Nov 2020) and only came to print this evening. Got exact same problem, part cooling fan not running but says 100% on display. I searched on the internet and found several threads on here with same problem. I clicked the button as someone mentioned and away it went.
So, will anyone at BTT re-compile the firmware with the fix please. Compiling my own is a bit beyoned me, not the compiling part but the mass of options that I just don't understand.
One good thing about this update, is that I can now do Mesh Bed Leveling which I could not do before on the last version, so don't want to lose that option by rolling back.

bleakreaper, 2020-11-21:
in my case i remove "//" from #define FAN_SOFT_PWM in configuration.h

dreisday, 2020-11-22:
I managed to overcome this by using the source from here: https://www.danbp.org/p/en/node/149
The only changes I made were to the BLTouch bed levelling, as I find 3 times in each spot takes far too long.

Since flashing this firmware both my cooling fan and case fan are working as expected. Elsewhere I've seen it suggested the behaviour is related to the `#define SPEAKER` function in configuration.h, and that is disabled in this firmware, so perhaps that's the fix.

bleakreaper, 2020-11-23:
> I managed to overcome this by using the source from here: https://www.danbp.org/p/en/node/149
> The only changes I made were to the BLTouch bed levelling, as I find 3 times in each spot takes far too long.
> 
> Since flashing this firmware both my part cooling fan and case fan are working as expected. Elsewhere I've seen it suggested the behaviour is related to the `#define SPEAKER` function in configuration.h, and that is disabled in this firmware, so perhaps that's the fix.

nope sorry i tried and still if i don't have FAN_SOFT_PWM active the fan only works on/off.

xtiang7c0, 2020-12-01:
Same issue for me. I do run an Ender 5 with BL Touch and used the "2.0.7.2 firmware-bltouch-for-z-homing.bin" firmware. Can someone share the pre-compiled files for this machine with me?

BigTreeTech SKR Mini E3 v2.0
BL Touch Smart Sensor

onebehler, 2020-12-14:
#459 Is what I feel I am seeing with this issue

HoaxOfClocks, 2021-01-02:
Has anyone figured out what is causing this? Still can't get my 5015 to reliably turn on on my Ender 3. I would try one of the configs linked in this thread but I don't use a BLtouch.

onebehler, 2021-01-02:
Are you editing your own firmware?  If so, I would try commenting out the #define Speaker as I referenced.  

If you are not editing your own firmware, I could see about helping you.  Leave a response here.  I will need a link to the firmware source you are basing your install on.

HoaxOfClocks, 2021-01-02:
> Are you editing your own firmware? If so, I would try commenting out the #define Speaker as I referenced.
> 
> If you are not editing your own firmware, I could see about helping you. Leave a response here. I will need a link to the firmware source you are basing your install on.

I am just starting out with editing firmware but yes, I have tried commenting out #define Speaker and also enabling soft PWM, as well as a few combinations of these in various versions of the firmware from both the Marlin and BTT github. However, I'm wondering if it may be an issue with hardware in my case. I tried swapping back to the stock 4010 blower and it worked fine, so I'm wondering if the 5015 I'm using is of low quality or drawing more current than the SKR can supply? Confirmed that the 5015 is 24v and wired correctly. The fan does shudder sometimes when attempting to start up, and might run for a few minutes at a time depending on conditions. Also I can feel magnetic resistance if I try to turn the fan by hand while it should be running.

For now, I've switched back to the stock 4010 with no issues. I'll have to experiment some more and maybe try a higher quality 5015?

onebehler, 2021-01-02:
It does sound like a bad 5015 fan in your case.  I have purchased several different manufacturers fans from Amazon and thankfully not had that problem yet.  Good luck

beeRacer505, 2021-01-07:
Hi, I have an Ender 3 Pro running the SKR mini E3 v2 and since I flashed Marlin 2.0.7 on it the part cooling fan isn't spinning anymore, regardless what I am trying to do. I ain't using BL-Touch and want to keep the Mesh BedLeveling Feature. Has anyone compiled yet a bugfix for this printer without bl-touch? I would glad if it could be shared so I can use it.

damy1976, 2021-01-09:
Sorry, i wrote basic english. me too have problem. i already new update firmware.. but always ON fan 0 and 1 both too.. cannot stop.. what abot it? i send show my video youtube: https://youtu.be/dDTbQYqRqps

thank you and best regards

autonumous, 2021-02-07:
have a look at my comments on https://github.com/MarlinFirmware/Marlin/issues/20670
Basically i’ve changed platformio settings for [common_stm32f1] to the original ones from BTT repo and not used the marlin defaults..

Now Fan is working as well as beeper and button (odd behaviour requiring double clicking for action to take effect)