bigtreetech/BIGTREETECH-SKR-mini-E3 #433: [BUG] SKR mini E3 V2.0 - Fan, stepper heat, no baby step and noise issues

RS231, 2020-10-24:
I upgraded from E3 mini V1.2 to V2.0 in my Ender 3 after the part cooling fan mosfet burnt. I really like the v1.2 as it worked well and was very very quite. 

After installing the V2.0 board I am having several issues. The main board fan does not switch on, ever. I believe it should go on as soon as the steppers draw current. I could not ge this to work, so connected the fan directly to 24v. I started printing an noticed the steppers are alot louder than the v1.2. Still less than the stock boards, but still a prominent stepper noise. Now I have noticed that the stepper motors are extremely hot, except the z stepper. I cannot touch the motor for more than 2 seconds. This was never an issue with the v1. 2, even when I printed in an enclosure. Especially the e stepper, it gets so hot the filament get soft in the gearing and constantly strips. I get major under extrusions and numerous failed prints. 

I have Tried adjusting the current to the motors from the default 800 down to 760 on the control panel (as recommended by "teachingtech") but I cannot notice a difference. 

I further cannot control baby steps as I was able to on the v1.2. To note I downloaded the marlin version from 2 or 3 weeks ago. 

**Expected behavior**
1. The main board fan should run off the main board when stepper motors switch on
2. The stepper drivers must be as quite a as the v1.2 if not better (otherwise it's a downgrade not an upgrade) 
3. I should be able to control z axisbbaby steps by double clicking the knob as I did on the v1. 2
4. The stepper motors should not run extremely hot. 

**Actual behavior**
1. Fan does not switch on
2. Still a very prominent stepper noise. V1.2 was silent. V2.0 is not silent 
3. No option in the menu to get to babysteps for z axis 
4. Stepper motor to hot to touch 

### Additional Information
Ender 3
Skr. Mini e3 V2.0 
BL touch used as probe
Marlin v7 with bltouch as probe.

brew99, 2020-10-27:
Are you using BTT marlin from this Github, or the Vanilla Marlin?   Go lower than 760mA.   Stock driver current on the V1.2 is 580mA (XYZ) and 650mA(E).   You cannot use the same firmware from V1.2 on the 2.0

PrutzerNL, 2020-11-01:
It looks like there is a bug.
When Stealthcop is enabled in the firmware, it is disabled.
Disable it, will enable it:

Run M569 S0 Z E X Y to disable stealthchop
Check M569 - shows stealthchop is on

RS231, 2020-11-13:
UPDATE:

Thank you for the input. Dropping the current that low caused the X stepper to skip. X, Y, Z is on 600 mA and E is on 650 mA.

So after struggling with it for a couple of weeks I resorted to loading the May version of the firmware.

1. I can adjust baby steps again. 
2. Steppers are quiter than before but not the same as v1.2. The stepper motors are running cooler now, which is good. 
3. Bl touch is now doing 24 level points which I don't like. 
3. For some reason I cannot set the probe offset lower. It does the bed level GCode, but as the print start I have to adjust the babystep everytime exactly 0.02 to get the correct first layer 
4. I have not tested whether the main board fan is working from the board

Still not happy with the v2.0 board. Too many issues still.