bigtreetech/BIGTREETECH-SKR-mini-E3 #435: [BUG] (short description)

PrutzerNL, 2020-10-26:
SKR Mini V 1.2 Marlin 2.7.2 has a bug.
When you change the Z offset the nozzle is not going down.

darthdively, 2020-10-27:
I'm looking for this also.

legend069, 2020-10-31:
i have same problem. i used main marlin tree though

legend069, 2020-11-01:
i think i found the problem.
inside config_advanced
look for "BABYSTEP_ZPROBE_OFFSET" and enable it.
i just found it and have not updated firmware yet since it's printing.

legend069, 2020-11-03:
confirmed it works

cedomir-spalevic, 2020-11-04:
@legend069 can you share the configuration files? I'm running into the same error

legend069, 2020-11-05:
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5496370/Marlin.zip)

there are my files i use.
please change any important stuff, and make sure you override your eeprom settings

RS231, 2020-11-13:
So I have the same issue on the V2.0 board. Any way to fix this with the control panel.

I have no experience with changing the config

cedomir-spalevic, 2020-11-13:
So I ended up changing my cables in order to get it to work:

1. I switched Z axis motor wires from ZAM to ZBM pin
2. I had to switch the black and white wires in my bltouch cables

Hope that helps. Thanks @legend069 for sharing the files