bigtreetech/BIGTREETECH-SKR-mini-E3 #436: SKR Mini E3 V2.0 Neopixel

zixoclock, 2020-10-26:
Hi, can someone please help me providing with Marlin 2.0.7.2 and Neopixel?  I tried my variations and mods, and i cannot get a compile bin for the SKR Mini E3 v2.  Can someone post the areas required to be changed on Marlin? Any help is appreciated. A step by step instruction will help.

abacabie, 2020-10-28:
There is a few steps involved but you could try the post linked below, this is what worked for me using the CommandoreBombardiero library with the skr mini v2.
[https://www.reddit.com/r/BIGTREETECH/comments/je7mrg/turn_off_neopixel_leds/g9or3eb/?context=3](url)

abacabie, 2020-10-28:
This is a copy of a post i left on Reddit and the only way i could get it to work with the skr mini v2.

Search for NEOPIXEL_LED in platformio.ini, you will find it under "Feature Dependencies" and overwrite "Adafruit NeoPixel@1.5.0" with "https://github.com/CommandoreBombardiero/Adafruit_NeoPixel"

when you compile this it will fail, go to the .pio folder in marlin and open the folder that downloaded the CommandoreBombardiero library, open the "Adafruit_NeoPixel.cpp" and find the line that reads "#include <replace with path to Delay.h file>" , its near the top, you need to overight and point to the "Delay.h" file that is in marlin, for me it looks like

"#include "C:\Users\ray\Desktop\Marlin-2.0.x\Marlin\src\HAL\shared\Delay.h"",

you will get a sanity check when you first compile, i used Ctrl and clicked the red warning to open the sanity check file and comment out so it looks like this

"#if ENABLED(NEOPIXEL_LED)

//#error "NEOPIXEL_LED (Adafruit NeoPixel) is not supported for HAL/STM32F1. Comment out this line to proceed at your own risk!"

#endif"

There maybe other ways of getting it to work but this was the only one that worked without any flickering or with colours been off, it also gave me Printer Event LEDs as well.

I have the "Configuration.h" file for neopixels set up like below, you will have to change the amount of LEDs for your own.

// Support for Adafruit NeoPixel LED driver

#define NEOPIXEL_LED

#if ENABLED(NEOPIXEL_LED)

#define NEOPIXEL_TYPE NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)

//#define NEOPIXEL_PIN 4 // LED driving pin

//#define NEOPIXEL2_TYPE NEOPIXEL_TYPE

//#define NEOPIXEL2_PIN 5

#define NEOPIXEL_PIXELS 30 // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)

//#define NEOPIXEL_IS_SEQUENTIAL // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.

#define NEOPIXEL_BRIGHTNESS 127 // Initial brightness (0-255)

#define NEOPIXEL_STARTUP_TEST // Cycle through colors at startup

zixoclock, 2020-10-28:
Success! Thanks man. You are the greatest.

abacabie, 2020-10-29:
> Success! Thanks man. You are the greatest.

Not the easiest instructions to follow so well done.

closed 2020-10-29 by zixoclock

dahose, 2020-10-30:
Hello community,

This is my first post in the forums, and I'm super excited I found this conversation.

I followed the instructions abacabie put in this thread, and I actually had mixed results.  I am working with an SKR MiniE3 V.2, Marlin 2.0.7.2, and a WS2812 24 LED Neopixel ring that I bought from Amazon.  My source files to Marlin were pulled from the BTT github repository.

My edited platformio.ini section looks as below.  You can see the original NEOPIXEL_LED lines are commented out, and I added the new ones.

_HAS_L64XX               = Arduino-L6470@0.8.0
                          src_filter=+<src/libs/L64XX> +<src/module/stepper/L64xx.cpp> +<src/gcode/feature/L6470>
#NEOPIXEL_LED            = Adafruit NeoPixel@1.5.0
**#                          src_filter=+<src/feature/leds/neopixel.cpp>
NEOPIXEL_LED            = https://github.com/CommandoreBombardiero/Adafruit_NeoPixel
                          src_filter=+<src/feature/leds/neopixel.cpp>**
MAX6675_IS_MAX31865     = Adafruit MAX31865 library@~1.1.0
USES_LIQUIDCRYSTAL      = LiquidCrystal@1.5.0_

My configuration.h looks like this.  If I used the type value of NEO_GRB the lights didn't work, so I tried NEO_GRBW, and the lights fired up.

_// Support for Adafruit NeoPixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRBW // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  // #define NEOPIXEL_PIN     4       // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 24       // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  //#define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup_

The path to my library directory looks like this - "C:\marlin\marlin_2.0.7.2_skr_mini_e3_v2\.pio\libdeps\STM32F103RC_btt_512K", and I saw that the "Adafruit_NeoPixel.cpp" file already had instructions on what to update.  Very cool.  So the start of my file looks like this.  I left the original include of Adafruit_Neopixel.h" in there, and my sanity check file already seemed to have the right lines commented out.

_* You should have received a copy of the GNU Lesser General Public
 * License along with NeoPixel. If not, see
 * <http://www.gnu.org/licenses/>.
 *
 */

#include "Adafruit_NeoPixel.h"

**//Locate Delay.h file in your local Marlin copy: Marlin 2.0/Marlin/Marlin/src/HAL/shared/Delay.h
//and enter full path to this file below

#include "C:\marlin\marlin_2.0.7.2_skr_mini_e3_v2\Marlin\src\HAL\shared\Delay.h"**

#ifdef TARGET_LPC1768
  #include <time.h>
#endif_

So my lights now work, and can be controlled my Octopi using the M150 plugin, or direct commands.  However, the led's are running in a variety of discrete colors.  I know from a previous configuration effort that my light CAN can do full white light on each one of them, so I will keep poking around to see if anyone comes up with a better solution that will let me actually put my light ring in full, white light mode.    One feature on the light ring that I might need to think about is an unused data output terminal.  For now, I am so very happy this thread exists, and I can now turn the lights on/off, which will allow for remote monitoring of my print jobs in the darkened office.

Thanks so much for your great infomation abacabie.

Jose

abacabie, 2020-10-30:
Glad you found it useful, the library i found from a post here but cant seem to find it again, its worth pointing out i use a 600ohm resistor and a 1000µF capacitor as well, these are with the standard 60 led WS2812B, i use a separate buck converter for the 5v supply.

theboys512, 2020-10-31:
> Glad you found it useful, the library i found from a post here but cant seem to find it again, its worth pointing out i use a 600ohm resistor and a 1000µF capacitor as well, these are with the standard 60 led WS2812B, i use a separate buck converter for the 5v supply.

Can I use the 5v extension that BTT offers? Also, May I know the voltage of the capacitor? Will it not work without the resistor and capacitor?

abacabie, 2020-10-31:
I believe the capacitor was 16v, BTT have the DCDC MODE V1.0 that seems to work well from what i have heard, i never tried it without the capacitor and resistor, it may well work fine without them, i have read plenty people not  using them without issue so you could try without first, it may be fine, adafruit have articles on neopixels that are interesting, may be worth a read.

theboys512, 2020-10-31:
> I believe the capacitor was 16v, BTT have the DCDC MODE V1.0 that seems to work well from what i have heard, i never tried it without the capacitor and resistor, it may well work fine without them, [...]

Thanks I'll check. Btw, 

 go to the .pio folder in marlin and open the folder that downloaded the CommandoreBombardiero library, open the "Adafruit_NeoPixel.cpp" and find the line that reads "#include " , its near the top, you need to overight and point to the "Delay.h" file that is in marlin, for me it looks like

"#include "C:\Users\ray\Desktop\Marlin-2.0.x\Marlin\src\HAL\shared\Delay.h"",

I checked mu pio folder but I couldn't find a Folder with the name CommandoreBombardiero. Could you please clarify what I need to do here please?

zixoclock, 2020-10-31:
theboys512,
here is the location for the Delay.h change.  Change path with your location
[image]

for resistor any value from 330 to 600ohm will do, the lower more LEDs.  For capacitor, 1000uF of any voltage above 10 will do.

theboys512, 2020-11-01:
@zixoclock thanks for your reply. The adafruit NeoPixel file is not in the path specified, which is why I was stuck. I downloaded it and copied it to the .pio directory and continued but I keep getting this error. 
Also, I noticed you have bugfix 2.0.7.2, I couldn't find that on marlinfw.org, the latest bugfix version says bugfix 2.0.x
[image]

zixoclock, 2020-11-01:
@theboys512, BTW, as shown, I renamed the my bugfix to folder 2.0.7.2.  Try to delete the .pio folder, this will create create and download all the necessary files again when compiling. Beyond that, I do not that to tell you.

androofoo, 2020-11-17:
> @zixoclock thanks for your reply. The adafruit NeoPixel file is not in the path specified, which is why I was stuck. I downloaded it and copied it to the .pio directory and continued but I keep [...]

I believe this is happening because it can't download the Github files. You need to download: https://git-scm.com/downloads/guis

theboys512, 2020-11-17:
@androofoo thanks it compiled. However, the startup sequence is all red, and the lights are off by default. And I tried the M150 R255 commands. Red and blue work fine, Green doesn't work. It turns on 3 of the 12 neopixels and it's red.
Do the colors work properly for you? With the startup test and all? 
These are my config settings;

#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN     PA8       // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 12       // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Support for second Adafruit NeoPixel LED driver controlled with M150 S1 ...
  //#define NEOPIXEL2_SEPARATE
  #if ENABLED(NEOPIXEL2_SEPARATE)
    #define NEOPIXEL2_PIXELS      15  // Number of LEDs in the second strip
    #define NEOPIXEL2_BRIGHTNESS 127  // Initial brightness (0-255)
    #define NEOPIXEL2_STARTUP_TEST    // Cycle through colors at startup
  #else
    //#define NEOPIXEL2_INSERIES      // Default behavior is NeoPixel 2 in parallel
  #endif

  // Use a single NeoPixel LED for static (background) lighting
  //#define NEOPIXEL_BKGD_LED_INDEX  0               // Index of the LED to use
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 } // R, G, B, W
#endif

luizmont, 2020-12-21:
First of all, thanks for all the solutions provided here.
I managed to build the firmware without problems, however my leds are not lighting up and when I try to do a M150 command, I get "Unknown command: "M150"".

Here is my settings from Configuration.h:

// Support for Adafruit NeoPixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  //#define NEOPIXEL_PIN     4       // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 10       // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  //#define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Support for second Adafruit NeoPixel LED driver controlled with M150 S1 ...
  //#define NEOPIXEL2_SEPARATE
  #if ENABLED(NEOPIXEL2_SEPARATE)
    #define NEOPIXEL2_PIXELS      15  // Number of LEDs in the second strip
    #define NEOPIXEL2_BRIGHTNESS 127  // Initial brightness (0-255)
    #define NEOPIXEL2_STARTUP_TEST    // Cycle through colors at startup
  #else
    //#define NEOPIXEL2_INSERIES      // Default behavior is NeoPixel 2 in parallel
  #endif

  // Use a single NeoPixel LED for static (background) lighting
  //#define NEOPIXEL_BKGD_LED_INDEX  0               // Index of the LED to use
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 } // R, G, B, W
#endif

Thanks!

LearnIIBurn, 2020-12-22:
> @androofoo thanks it compiled. However, the startup sequence is all red, and the lights are off by default. And I tried the M150 R255 commands. Red and blue work fine, Green doesn't work. It turns on 3 of the 12 neopixels and it's red.
> Do the colors work properly for you? With the startup test and all?
> 
> I tried setting the Neopixel pin to PA8, 4, and I tried commenting it out as well. No luck.
> These are my config settings;
> 
> #define NEOPIXEL_LED
> #if ENABLED(NEOPIXEL_LED)
> #define NEOPIXEL_TYPE NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
> //#define NEOPIXEL_PIN 4 // LED driving pin
> //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
> //#define NEOPIXEL2_PIN 5
> #define NEOPIXEL_PIXELS 12 // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
> #define NEOPIXEL_IS_SEQUENTIAL // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
> #define NEOPIXEL_BRIGHTNESS 127 // Initial brightness (0-255)
> #define NEOPIXEL_STARTUP_TEST // Cycle through colors at startup
> 
> // Support for second Adafruit NeoPixel LED driver controlled with M150 S1 ...
> //#define NEOPIXEL2_SEPARATE
> #if ENABLED(NEOPIXEL2_SEPARATE)
> #define NEOPIXEL2_PIXELS 15 // Number of LEDs in the second strip
> #define NEOPIXEL2_BRIGHTNESS 127 // Initial brightness (0-255)
> #define NEOPIXEL2_STARTUP_TEST // Cycle through colors at startup
> #else
> //#define NEOPIXEL2_INSERIES // Default behavior is NeoPixel 2 in parallel
> #endif
> 
> // Use a single NeoPixel LED for static (background) lighting
> //#define NEOPIXEL_BKGD_LED_INDEX 0 // Index of the LED to use
> //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 } // R, G, B, W
> #endif

Sorry kinda late responding to your issue but I believe I had the same behavior and figured it out. How are you wired up with your neopixels? I actually grabbed a small breadboard since the resistor and capacitor are in play and wired everything up at the breadboard with a 5v power supply powering the LEDs vs. the SKR board doing it. The only thing the SKR board is doing is data to the LEDs via the PA8 pin (at first).

Then I had the weird color off issues like you and remembered that you have to have a common ground between the controller and the neopixels, so I ran a jumper from the neopixel negative on the SKR to the negative bus on the breadboard and everything is good.

Maybe that will help your issue depending on how you are wired up.

ZixZeven, 2020-12-22:
I would like to add the following. The '#define NEOPIXEL_LED' shown is good and I am using Marlin BugFix dated Nov30 . No other changes on Marlin.  I am driving 12 LEDs, sending M150 working OK and colors OK too.  I am using a 470ohm resistor on Signal and 1000uF/16V on the 5V.

gazzaman2k, 2021-01-10:
works great on 2.0.7.2 as of 10 jan 2021

only had to do this bit:

Search for NEOPIXEL_LED in platformio.ini, you will find it under "Feature Dependencies" and overwrite "Adafruit NeoPixel@1.5.0" with "https://github.com/CommandoreBombardiero/Adafruit_NeoPixel"

when you compile this it will fail, go to the .pio folder in marlin and open the folder that downloaded the CommandoreBombardiero library, open the "Adafruit_NeoPixel.cpp" and find the line that reads "#include " , its near the top, you need to overight and point to the "Delay.h" file that is in marlin, for me it looks like

"#include "C:\Users\ray\Desktop\Marlin-2.0.x\Marlin\src\HAL\shared\Delay.h"",

the insanity check part was not required, this is for just 1 neopixel running from the header PA8 onboard the e3 v2

sbradley88, 2021-01-20:
Hi there. Thanks for taking the time to sort this problem out with people. I seem to have another issue I cannot work out, if you don't mind helping? I'm good at following good instructions however this one is totally throwing me. VSCode failing to compile and is asking me about GPIO sets and stuff and i have no idea what to do about them

_.pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp: In member function 'void Adafruit_NeoPixel::show()':
.pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1873:9: error: 'gpio_set' was not declared in this scope; did you mean 'LPC176x::gpio_set'?
 1873 |         gpio_set(pin);
      |         ^~~~~~~~
      |         LPC176x::gpio_set
In file included from C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/pinmapping.h:21,
                 from C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:36,
                 from .pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.h:49,
                 from .pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:46:
C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/gpio.h:44:36: note: 'LPC176x::gpio_set' declared here
   44 | [[gnu::always_inline]] inline void gpio_set(const pin_t pin, const bool value) {
      |                                    ^~~~~~~~
.pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1874:9: error: 'time' has not been declared
 1874 |         time::delay_ns(550);
      |         ^~~~
.pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1876:9: error: 'gpio_clear' was not declared in this scope; did you mean 'LPC176x::gpio_clear'?
 1876 |         gpio_clear(pin);
      |         ^~~~~~~~~~
      |         LPC176x::gpio_clear
In file included from C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/pinmapping.h:21,
                 from C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:36,
                 from .pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.h:49,
                 from .pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:46:
C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/gpio.h:28:36: note: 'LPC176x::gpio_clear' declared here
   28 | [[gnu::always_inline]] inline void gpio_clear(const pin_t pin) {
      |                                    ^~~~~~~~~~
.pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1877:9: error: 'time' has not been declared
 1877 |         time::delay_ns(450);
      |         ^~~~
.pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1881:9: error: 'gpio_set' was not declared in this scope; did you mean 'LPC176x::gpio_set'?
 1881 |         gpio_set(pin);
      |         ^~~~~~~~
      |         LPC176x::gpio_set
In file included from C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/pinmapping.h:21,
                 from C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:36,
                 from .pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.h:49,
                 from .pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:46:
C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/gpio.h:44:36: note: 'LPC176x::gpio_set' declared here
      |                                    ^~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\LPC1769\libf48\Adafruit NeoPixel\Adafruit_NeoPixel.cpp.o] Error 1
======================================================================================================== [FAILED] Took 34.49 seconds_ 

Any information or help you can provide will be really appreciated!

gazzaman2k, 2021-01-20:
you seem to have the LPC1769 chosen your board is a LPC1768

it should look like this in platform.io tab

[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = LPC1768
include_dir  = Marlin

sbradley88, 2021-01-20:
> you seem to have the LPC1769 chosen your board is a LPC1768
> 
> it should look like this in platform.io tab
> 
> [platformio]
> src_dir = Marlin
> boards_dir = buildroot/share/PlatformIO/boards
> default_envs = LPC1768
> include_dir = Marlin

I have tried this but i get the same error messages

_.pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp: In member function 'void Adafruit_NeoPixel::show()':
.pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1873:9: error: 'gpio_set' was not declared in this scope; did you mean 'LPC176x::gpio_set'?
 1873 |         gpio_set(pin);
      |         ^~~~~~~~
      |         LPC176x::gpio_set
In file included from C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/pinmapping.h:21,
                 from C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:36,
                 from .pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.h:49,
                 from .pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:46:
C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/gpio.h:44:36: note: 'LPC176x::gpio_set' declared here
   44 | [[gnu::always_inline]] inline void gpio_set(const pin_t pin, const bool value) {
      |                                    ^~~~~~~~
.pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1874:9: error: 'time' has not been declared
 1874 |         time::delay_ns(550);
      |         ^~~~
.pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1876:9: error: 'gpio_clear' was not declared in this scope; did you mean 'LPC176x::gpio_clear'?
 1876 |         gpio_clear(pin);
      |         ^~~~~~~~~~
      |         LPC176x::gpio_clear
In file included from C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/pinmapping.h:21,
                 from C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:36,
                 from .pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.h:49,
                 from .pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:46:
C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/gpio.h:28:36: note: 'LPC176x::gpio_clear' declared here
   28 | [[gnu::always_inline]] inline void gpio_clear(const pin_t pin) {
      |                                    ^~~~~~~~~~
.pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1877:9: error: 'time' has not been declared
 1877 |         time::delay_ns(450);
      |         ^~~~
.pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:1881:9: error: 'gpio_set' was not declared in this scope; did you mean 'LPC176x::gpio_set'?
 1881 |         gpio_set(pin);
      |         ^~~~~~~~
      |         LPC176x::gpio_set
In file included from C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/pinmapping.h:21,
                 from C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\cores\arduino/Arduino.h:36,
                 from .pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.h:49,
                 from .pio\libdeps\LPC1769\Adafruit NeoPixel\Adafruit_NeoPixel.cpp:46:
C:\users\sbrad\.platformio\packages\framework-arduino-lpc176x\system\lpc176x/gpio.h:44:36: note: 'LPC176x::gpio_set' declared here

Environment    Status    Duration
-------------  --------  ------------
LPC1769        FAILED    00:04:33.056
=================================================================================================== 1 failed, 0 succeeded in 00:04:33.056 ===================================================================================================PS D:\Users\Scott\Downloads\BIGTREETECH-SKR-E3-Turbo-master\BIGTREETECH-SKR-E3-Turbo-master\firmware\Marlin-2.0.7.2-SKR-E3-Turbo> echo "done" >C:\Users\sbrad\AppData\Local\Temp\ipc
PS D:\Users\Scott\Downloads\BIGTREETECH-SKR-E3-Turbo-master\BIGTREETECH-SKR-E3-Turbo-master\firmware\Marlin-2.0.7.2-SKR-E3-Turbo>_

I wonder if it is another change ive made causing this

gazzaman2k, 2021-01-20:
is it a e3 turbo or a e3 v2 you have? the turbo has a different chip on it same as the skr 1.4turbo if its a turbo then you need LPC1769

sbradley88, 2021-01-20:
It is the E3 Turbo. I had it working fine before trying to add these neopixels