bigtreetech/BIGTREETECH-SKR-mini-E3 #439: [BUG] Straight PWM destroys brushless DC fan motors.

sumguysr, 2020-10-30:
Brushless DC fans contain sensitive driver electronics which are not designed to stand stand up to a high frequency square wave input. The fan output should have an additional MOSFET and a capacitor to make a safe half-bridge motor controller. No doubt that would also solve the many problems people are having with some fans not working, some fans only working at 100%, early fan death, and annoying PWM noise.

1CM69, 2021-01-28:
This must be the issue I am having too.
3 part cooling fans burnt out in 3 days when testing bridging settings, turns out that anything under 100% speed for any length of time kills the fan.

hitokiri8x, 2021-01-29:
This could be because my new fan 5015 broken after few prints...
What are your suggestion to fix/mitigate or fan to buy?

1CM69, 2021-01-29:
I've been doing some research and the overwhelming result is that the fan needs to be brushed. 
That's a real problem as it is impossible to find a useable brushed 24v fan for these installations. 
At the moment I get by by running my fan at 100% or 0% only, anything in between kills it. 
I find it hard to believe that the same make & model of fan that I use now, I was using in my OE Creality board without any issues any any percentage but this board destroys them. 

I am still researching this more as most RC models use brushless motors & control speed via an ESC. Perhaps there is some difference in the PWM functionality of an ESC compared to the PWM on this board, someone with more knowledge might add comment here.

hitokiri8x, 2021-01-31:
This is a problem..
For your knowledge is the same for the default stock 3 board? 
I found a fan on amazon that a guy says it works well with his ender 3 and petsfang duct...

Getting another 5015 and running  it on/off seems a lot of air, even tought I never tried it.