bigtreetech/BIGTREETECH-SKR-mini-E3 #442: SKR V2.0+BLTouch= No centering!!

FPVScrub, 2020-10-31:
I have an Ender 3 Pro with the SKR v2.0 board. I just installed a genuine BLTouch and since I didn't connect the wiring to the Z-Port, I'm using the Firmware-BLTouch.bin file. Everything powers up just fine and I am able to perform a "level bed" and store the setting. When I perform an Auto Home, the extruder head stays at the at the corner and never centers to the center of the bed so I can perform a Z-offset. I tried installing the "replaces z-switch" firmware ,which should be incorrect given the board connections I used, but the head will go to center under auto home. Once centered, there is an error displayed stating something along the lines of "too far." Need advice here please. If I were proficient at compiling MArlin I would simply do so.

legend069, 2020-11-03:
i have same board and clone bltouch.
my hotend setup is different, you can look at my firmware if you like?
mine is from the main marlin tree

FPVScrub, 2020-11-04:
> i have same board and clone bltouch.
> my hotend setup is different, you can look at my firmware if you like?
> mine is from the main marlin tree

Thank you for your reply. At this point I have given up trying to get help from BigTree, literally ZERO response. I have simply resumed printing without the use of BLTouch since I have way too much projects I need to get going on. My friend will eventually compile a fresh version of Marlin for me then I will report back. As of now, even if everything works out, I will never buy another of this companies products. They simply rely on the community to provide customer service on their behalf.

madeinholland, 2020-11-05:
i have the same problem

FPVScrub, 2020-11-05:
> i have the same problem

Well I have news for you, you're screwed unless you can find someone to compile a fresh copy of Marlin for you to try. My friend that has a tad more experience than me tried compiling a fresh copy of Marlin but the file is missing the "platformio.ini" file so the file can't be modified in PlatformIO. 

Also, the "resume print after power loss" doesn't work (ask me how I know) and the filament cooling fan ONLY works at 100%. That means if you're trying to print PETG at 40% fan setting, the fan refuses to run. Way too many issues and BTT is impossible to get ahold of. They rely on the community providing support. My friend has recommended I send this board back and pick up a TH3D board instead. They have 1000% real customer support.

Let this be a lesson to everyone: Youtubers recommend these products because they get paid to and they also receive direct support but for you, you're a schmuck so expect nothing.

legend069, 2020-11-05:
the "print after powerloss" can mess up the printer/sd card you don't want that

if you're scared of a power cut during a long print buy  UPS
 your fan problem is related to #define FAN_SOFT_PWM

[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5496378/Marlin.zip)
have a look at my firmware files.

FPVScrub, 2020-11-05:
> the "print after powerloss" can mess up the printer/sd card you don't want that
>
> if you're scared of a power cut during a long print buy UPS
> your fan problem is related to #define FAN_SOFT_PWM [...]

I'll forward that zip to my friend so he can have a look at it when he gets the chance. When I first installed my SKR Mini, it had the "print after powerloss" enabled. I found out when I actually had a power loss and was able to resume my print. I lost that feature when I installed the BLTouch firmware which, along with the other BTT firmware variants, had been updated. Since I didn't save the original bugfix firmware my board came with, I couldnt reinstall it. The fan thing has been an issue from the day I installed the board and as I've mentioned, I don't have a lot of experience with Marlin but thank you very much for the information. 

I'm convinced this is an improperly compiled firmware issue and I wish I had the time to dedicate to learning coding Marlin however, I should still be able to get help from BTT and from reading around, almost no one does.

depawlur, 2020-11-06:
had same issue
@legend069, how you have you bltouch plugged? plug it properly into Z-port plug on the motherboard as described in the manual.

also make sure it's FIRM because for me whatever shitty cables were coming with BLTouch extender were loose.

legend069, 2020-11-08:
i had to swap wires around(JST style connector ), still using the same z endstop plug

madeinholland, 2020-12-04:
I have compiled  the firmware but I want to Home with my BLtouch ,
But what every I do with z homing it stays ideal in the corner. How can I fix this.
ore can somebody share his .h and .ahv files  

I have an skr mini v2.0 board on my ender 3 pro. I get the firmware working without use off the bltouch