bigtreetech/BIGTREETECH-SKR-mini-E3 #443: Configuration files need

Jedrusia, 2020-11-01:
I have an Ender 3 Pro with SKR mini E2 V2 motherboard and LCD TFT 35v 3.0. I also bought and installed BLTouch (connected to ZPROBE), Smart Filament sensor, BTT UPS 24V V1.0 and BIGTREETECH V1.2 relay, power monitoring module.
I have installed the latest firmware 2.0.7.2 firmware-bltouch.bin:
I started the board and BTT UPS turns off the power after some time. I know how to solve this problem by changing the firmware. The BLTouch module itself works very well with this firmware. The problem is elsewhere. To rip all the elements together, I would need complete firmware configuration files, in which BLTouch (via ZPROBE) Smart Filament sensor, BTT UPS 24V V1.0 and BIGTREETECH V1.2 relay power monitoring module are already enabled (in HIGH output).
The most important thing for me is to get the full configuration of all the necessary files with BLTouch settings. The settings from the downloaded foirmware are very good under BLTouch, but I miss the source files. Does it somehow set in firmare: BTT UPS 24V V1.0 and Smart Filament sensor? I have nowhere found how to do this.
Please help.

kiwipaulrob, 2020-11-22:
There is a custom config file for the 1.2 version board  as part of the merlin download that you can start with.

https://github.com/MarlinFirmware/Configurations/tree/release-2.0.5/config/examples/BigTreeTech/SKR%20Mini%20E3%201.2

see this guideline

https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/

Jedrusia, 2020-11-22:
Thanks.

I was resolving my problem alone

 

https://github.com/MarlinFirmware/Configurations/tree/release-2.0.5/config/examples/BigTreeTech/SKR%20Mini%20E3%201.2

https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/

kiwipaulrob, 2020-11-22:
The reddit author also has a 2.0 commentary / guide.Sent from my Galaxy

 

 

https://github.com/MarlinFirmware/Configurations/tree/release-2.0.5/config/examples/BigTreeTech/SKR%20Mini%20E3%201.2

https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/