bigtreetech/BIGTREETECH-SKR-mini-E3 #444: SKR MINI E3 V2.0[BUG] (Power loss

frankcannon, 2020-11-03:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**
 to resume after power loss

**Actual behavior**
go to the point of resuming but it won't move

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

slmalb, 2020-11-11:
Yes same problem with E3 V2.0 with ups nor bare sdcard could not start the print. 
on screen says "Printing" but not starts to print.
Marlin 2.0.7 bf

akurekin, 2020-12-05:
Same problem

slmalb, 2020-12-14:
Solved with the new bugfix of Marlin. Try to edit and compile from the scratch. No solition for mini ups but without it it finaly starts to print after power loss

atharvbadkas, 2020-12-15:
> Solved with the new bugfix of Marlin. Try to edit and compile from the scratch. No solution for mini ups but without it it finaly starts to print after power loss

Good,
If there's a solution with btt ups can you let us known in this issue thread?? 
@slmalb

atharvbadkas, 2020-12-18:
Hi...I contacted BTT And they sent me files claiming they have tested it on BTT SKR MINI E3V2 AND BTT UPS 24V for Ender 3..but I'm still facing same issue with their firmware also....can u test it on your machine...and let me know??.... thanks...link https://drive.google.com/folderview?id=1lLJ8wB38pCyJNlhxEXdSRXJNUCHG2Eis

slmalb, 2020-12-18:
I havent an Ender 3 but still BTT UPS problem that I have exist. 
Also the file is not accessible.

atharvbadkas, 2020-12-18:
> I havent got an Ender 3 but still BTT UPS problem that I have exist.
> By the way your file is not accessible.

Check it now...i have made tome changes

akurekin, 2020-12-18:
The same problem is described in https://github.com/MarlinFirmware/Marlin/issues/19602
The solution would be to disable interrupted_file_exists() function in powerloss.h that does not work properly:

// The referenced file exists
    //static inline bool interrupted_file_exists() { return card.fileExists(info.sd_filename); }
    //static inline bool valid() { return info.valid() && interrupted_file_exists(); }
    static inline bool valid() { return info.valid(); }

javicarrera, 2021-01-24:
Same here with SKR 1.4 and Mini UPS v2.0:
Just send to print from SD, goes Z up a little and get the Power loss recovery,
Have tried the changes in powerloss.h and still does the same

RECV: echo:SD card ok
SENT: M105
RECV: ok T:21.95 /0.00 @:0
SENT: M105
RECV: ok T:21.95 /0.00 @:0
RECV: echo:Now fresh file: 1xflat~1.gco
RECV: File opened: 1xflat~1.gco Size: 620289
RECV: File selected
RECV: ok
RECV: //action:resume
RECV: ok
RECV: Write Job Recovery Info...
RECV: valid_head:1 valid_foot:1
RECV: current_position: -7.0000,-19.0000,3.0000,0.0000
RECV: zraise: 2.0000
RECV: home_offset: 0.0000,0.0000,0.0000
RECV: position_shift: 0.0000,0.0000,0.0000
RECV: feedrate: 100
RECV: target_temperature: 0
RECV: fan_speed: 0,0
RECV: leveling: 0 fade: 10.0000
RECV: sd_filename: /1XFLAT~1.GCO
RECV: sdpos: 0
RECV: print_job_elapsed: 0
RECV: dryrun: 0
RECV: allow_cold_extrusion: 0
RECV: ---
RECV: Writing to file: /PLR
RECV: echo: cold extrusion prevented
RECV: Error:Printer halted. kill() called!
[ERROR] Error:Printer halted. kill() called!

RECV: //action:poweroff

matthewross257, 2021-03-29:
Ender 3 Pro, BTT SKR E3 v2 board. I have tried both of the Marlin 2.0 and mine will ask you to resume print, heat up, move to the last position, and just sit there. That has been the general result of all my tries.