bigtreetech/BIGTREETECH-SKR-mini-E3 #446: [BUG] filament change error marlin 2.0.7.2

redgod, 2020-11-06:
### Description

### Steps to reproduce

1. change of filament during printing
2. retraction and bleeding of the filament well
3. does not remember the apposition of origin

**Expected behavior**
I expected a perfect filament change

**Actual behavior**
After purging, the nozzle does not remember the correct height and position, since it begins to print more in the center and leaving the external wall exposed without filament

[image]

[image]

[image]

ysoline, 2020-12-13:
I have similar problem :(

Bastlwastl84, 2020-12-16:
you can edit at Filament Setting the unload and load lenth. perhaps a litttle bit more. did it had purged Filament before restart Print?