bigtreetech/BIGTREETECH-SKR-mini-E3 #447: Marlin-2.0.7.2-SKR-mini-E3-V2.0 [BUG] (M106 - Set Fan Speed stops working once )

depawlur, 2020-11-07:
Hi, I'm getting strange behavior on my nozzle fan.
M106 - Set Fan Speed command only works BEFORE print but doesn't do anything while printing or after the print.

so if I turn printer on, then issue e.g.
M106 S255 (or M106 P0 S255) 

it turns off I can modify the speed and what not -- all good

If I kick off a print gcode with say, 0 fan speed then somewhere after 4 layers I turn it on to 255 --- Fan doesn't start, print finishes and even if I try to start it -- M106 S255 doesn't do anything.

same in the opposite, if I start a print with 255, when print finishes, I can't turn off the fan or control the speed...

super weird issue

Please let me know if anyone could please help me debug.

thank you

bleakreaper, 2020-11-07:
in my case i remove "//" from #define FAN_SOFT_PWM in configuration.h

depawlur, 2020-11-09:
Hey, yeah. 
@bleakreaper thinks for the tip. That DID the trick. 

is it some sort of a bug? Because it did work for me until I started the print, I could also change the speed, didn't try clicking on a knob but still. 

Works as expected now.
Thanks again!

onebehler, 2020-12-14:
#459 is what I am feeling I am seeing with this issue

rcxrbt, 2020-12-31:
Where i connected the fan? FAN1 or FAN0?

davidrichter-cz, 2021-01-22:
Hi, I have the same problem with the blower fan, I do not have a BL touch, the problem occurred by changing the default display to TFT24. So I don't believe there is a bug on the E3 MINI board but in the firmware. Any advice?

DarkSnart, 2021-02-13:
@davidrichter-cz similar issue here. I've been all over the config files, pins files, github and the rest of the solar system searching for answers. Could be MOSFET failure as it looks like they're pretty weak from BTT and seem to be prone to failure. In my case, part cooling fan runs constantly and cannot shut it off with M107 nor via control interface. Display shows 0% fan, yet it's running, and when I turn it up to 100% it slows down but shows on the display as running faster! I just spent 7 hours between interrogating the definition files and compiling over 19 different versions of FW, and nothing has helped my particular issue. I read one post stating the person uses one version when he needs the fan on, and flashes a different version when he needs it off! Crazy hey? I'll post more if I learn anything new.

@rcxrbt on a V2.0 board the CPU fan(s) connect to FAN1 and is only run when the steppers are active. The part cooling fan connects to FAN0 and should be completely controllable through the display interface as well as M106/M107 commands.