bigtreetech/BIGTREETECH-SKR-mini-E3 #454: FAN0 always on - ignores gcode abs marlin settings

nikon72, 2020-11-15:
Firmware 2.0.7.2

FAN0 always on from power up (part cooling fan)

Doesn’t respond to gcode or the settings via the tft display 

V2 of the board 

Stock ender 3 pro, no additional accessories

flavio-costa, 2020-11-17:
same problem with me, only use abs, same profile configuration and since the last update the nozzle fan does not respond to the settings of gcode, tft or marlin simulator .... fan works whenever, when it works.

flavio-costa, 2020-11-17:
if you put the sd card directly on the board, fan0 works perfectly respecting the settings of the gcode, but if you put it in the tft inputs, fan0 does not respect the settings of the gcode. and even changing the display fan0 does not work.
it must be something from this last update of the display firmware (use tft24 v1.1)

Nobodywasfaster, 2020-11-27:
Same behavior with me "stock ender 3" and SKR Mini E3 V2 with version Firmware 2.0.7.2

davidrichter-cz, 2021-01-22:
Hi, I have the same problem with the blower fan, I do not have a BL touch, the problem occurred by changing the default display to TFT24. So I don't believe there is a bug on the E3 MINI board but in the firmware. Any advice?

Nobodywasfaster, 2021-01-24:
Hi,

i did a software downgrade to the original skr version 2.0.0.x. Since the downgrade the fan is working well.

But now the "break" feature (and the filament-change feature) did not work.

Both times the printer will no resume the print.....

 

In the next few days i will take some test with Octoprint / octpi. I hope with octoprint i could

use the fileament-change feature.....

 

 

Gesendet: Freitag, 22. Januar 2021 um 12:15 Uhr

DarkSnart, 2021-02-07:
Been on the internet for days trying to figure this out. This is an issue with a lengthy history and many workarounds (and I've tried every one so far), yet no solid fix.  Seems every time the issue is addressed, the next update breaks it again. I've read suggestions anywhere from swapping the fan wires to totally re-pinning the board. Maybe the Mini-E3 V3.0 will fix it once and for all. Until then I'll follow along and hope someone comes up with something!

pa1ntedb1rd, 2021-02-19:
Same problem here after 2 months of usage. Requested support from BTT via aliexpress, and they asked me to;
 "If so, please test the continuity of fan0's MOSFET to check whether it has a short circuit problem. Please give us a high-resolution picture of fan0's MOSFET by the way, I need to confirm its model."
I provided pictures and will do the connectivity check and share it. Not sure what will be the resolution. I will keep you posted.