bigtreetech/BIGTREETECH-SKR-mini-E3 #455: [BUG] No Extruder Movement

DarkPand0r, 2020-11-15:
Not sure this is a bug or an issue with the board, But I'm getting no extruder motor movement using the Ender 3 with BLTouch for Z Homing Firmware, I've plugged the Extruder into the Z Motor Header & it moves perfectly, Same if I plug the X Motor cable into the Extruder motor it works however I get no response from the Extruder Motor plug on the board even trying the Z axis motor through it.

DarkPand0r, 2020-11-15:
Please ignore and delete this... I'm a spoon and wish to hide my stupid.

Andearen42, 2022-02-05:
Considering I have the same problem, I would love to know how you fixed this, especially if its actually stupid easy. I've checked my firmware, wiring, swapping motors, and the E port energizes the motor but will not move.