bigtreetech/BIGTREETECH-SKR-mini-E3 #457: M300 Tones Extremely Harsh After Tone Fix

yourbuddydinec, 2020-11-29:
After following steps outlined [Here](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/241#issuecomment-639149248) the speaker/beeper on my Ender 3's screen sounds EXTREMELY harsh like nails on a chalk board. The beeps also seem to play much longer than they had before using the same M300 commands before enabling speaker.

This is on the Skr Mini E3 v1.2 with Marlin-bugfix-2.0.x from 9/24/2020

### Steps to reproduce

1. Enable #define SPEAKER in marlin
2. Delete framework-arduinoststm32-maple folder
3. Compile so it will give me new tone.cpp
4. Edit tone.cpp so that line 34 is now #define TONE_CHANNEL 4
5. Compile in VSCode and flash firmware
6. Send M300 command through octoprint or through gcode on sd card.

**Expected behavior**
Normal sounding beep but able to produce different tones and for requested length

**Actual behavior**
Able to produce different tones but extremely harsh to ear compared to before as well as longer tones than supposed to be

[Pastebin of Configuration.h](https://pastebin.com/0s9Bj4B2)
[Example of sounds. This is the command M300 S660 P1000 so should only be 1 second long](https://youtu.be/J4fBUILU5BU)

Shine753, 2020-12-24:
Same problem here with **Skr Mini E3 v2.0** with **Marlin-bugfix-2.0.x** from **28 Nov. 2020**

Cloud-Galaxyman, 2021-10-21:
Same here. Some tones sound fine, others sound like terrible chirps. That's too bad. I wrote a custom little song for when it's done  with a print. 

I have an Ender 3, SKR Mini E3 1.2 and Speaker is enabled.