bigtreetech/BIGTREETECH-SKR-mini-E3 #459: [BUG] Enabling #define SPEAKER prevents part cooling fan from working correctly

Valerious-Black, 2020-11-30:
Currently using "Marlin-2.0.7.2-SKR-mini-E3-V2.0" from right here. Everything works as expected but when i enable #define SPEAKER although the buzzer on the stock display does work, this prevents the part cooling fan from working properly.

With the speaker option defined, the part cooling fan operates in either off or full speed mode. 0-98% it is off, at 99% and 100% it spins up to maximum (using a single BTT 5015 blower style fan). I have tried flashing several times and the bug is consistent, if speaker is defined, fan not functioning properly, if speaker is not defined the fan works perfectly.

Steps to reproduce

1. enable #define SPEAKER in Configuration.h Line 1810
2. Compile and flash firmware
3. Enter menu, navigate to temperature and change fan percentage

**Expected behavior**
Speaker plays different tones, part cooling fan speed is variable and can be controlled from 0% to 100%

**Actual behavior**
Speaker plays different tones, part cooling fan is off between 0% and 98% and then full speed at 99% and 100%.

onebehler, 2020-12-14:
I can confirm similar results with same BTT firmware on stock parts cooling fan.  Comment out #define SPEAKER with no noticeable change to display sounds and parts cooling operates 30-100%.  

As delivered firmware came with #define SPEAKER not commented out to the best of my knowledge.

GJSchaller, 2021-07-31:
It looks like there's a fix from the main Marlin project:

https://github.com/MarlinFirmware/Marlin/issues/20670