bigtreetech/BIGTREETECH-SKR-mini-E3 #460: [BUG] Bed Leveling issue on Ender 3 v2 with BL Touch

thatsumoguy, 2020-11-30:
### Description

I keep having an issue where when I go to print it is clear that the BL Touch is not changing the Z level and some parts are great, other parts are too close, and other parts are scraping the bed. I even use the bed leveling visualizer on Octoprint (see attachment for pic) and the bed is fairly level (the most out of whack is 0.21 high) so I am not sure what is happening. I have also reflashed the firmware a couple times with a fresh download each time.

### Steps to reproduce

1. Run ABL, don't get a pop up asking to save the data, but I go ahead and run a M500.
2. Run G29 and M500 after my prints run a G28.

**Expected behavior**
For it to print level.

**Actual behavior**
It is not printing level.

[image]

madeinholland, 2020-12-31:
which board are you using

thatsumoguy, 2020-12-31:
> which board are you using

Mini E3 V2

stabtrimout, 2021-09-23:
How did you fix that problem? I have same issue!

Nerdtality, 2021-11-14:
Modify the firmware using latest bugfix and the example file, mine works just fine.