bigtreetech/BIGTREETECH-SKR-mini-E3 #462: [BUG] (Auto bed leveling not working)

Florecs71, 2020-12-02:
### Description

I Installed an BTT SKR MINI E3 V1.2 and the BL on a Ender 3 but when I try to do a Z Homing the Z axis is far apart from the bed and the BLT activate 3 times, and then appear a M999 BLtouch error . So I check the wiring and its good. I uploaded the "firmware-bltouch-for-z-homing.bin" and the problem still appears. So I configured the Marlin again and uploaded but the error appears again. Any ideas what could be the problem?

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

**Actual behavior**

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

soldop, 2020-12-15:
Having this very same issue.  Thought maybe my wire adaption I had to do was not making contact, but now hearing you've checked, I'm wondering if its something in my firmware settings.  Have you yet discovered the issue? work around?

joshfedo, 2020-12-17:
I had the same issue these are the things i had to check and change:

Did you rename the firmware to "firmware.bin"?
Is your probe hooked up to the probe/servo pins? (Not the z end stop header)
Is you z end stop installed and plugged into the bord?

If you removed your z endstop i believe you have to make some firmware changes. I just reinstalled my end stop and it working fine for me.

soldop, 2020-12-17:
Thanks for the feedback.  I guess it would make sense for me to make sure my pins are working as they are plugged in to the probe/server.  But fit was questionable.  
I did rename the file and did not remove the z end stop.

Florecs71, 2020-12-18:
I switched to "mesh_bed_leveling" and I've been working like this for 5 days and it works perfectly, but I also want to fix the problem with BLtoch. Did you succeed?

SjamonDaal, 2020-12-30:
> I switched to "mesh_bed_leveling" and I've been working like this for 5 days and it works perfectly, but I also want to fix the problem with BLtoch. Did you succeed?

I did the same thing, after troubleshooting and lots of searching I decided to buy a new BLTouch and the v2.0 board but this did NOT resolve my issues.

Florecs71, 2020-12-30:
> I did the same thing, after troubleshooting and lots of searching I decided to buy a new BLTouch and the v2.0 board but this did NOT resolve my issues.

I finally chose "mesh_bed_leveling" and it works perfectly. Try...

SjamonDaal, 2020-12-30:
> I finally chose "mesh_bed_leveling" and it works perfectly. Try...

I did try "mesh_bed_leveling". Then you have to probe (9) points manually with a peace of paper. But I still want to use my BLTouch.