bigtreetech/BIGTREETECH-SKR-mini-E3 #464: After firmware upgrade, tmc connection error.

Emal011, 2020-12-03:
Hi,

I tried to upgrade my SKR-mini-E3 v1.2 to the latest version from this site. After loading the firmaware as here described, I get an error TMC Connection Error. I searched the Internet for help, some TechSites recommended to build your own firmaware, I did it, but the error still there. 
I can move the Y and Z Axe, but if I try the X-Axe the printer sow an error, process kill()
So, after trying to figure out, what this error means, I connected my printer to my laptop and try to send some commands to read the values saved in the printer, the command m122 show me this:

```
>>> m122

SENDING:M122

	`X	Y	Z	E`

Address 0 0 0 0

Enabled false false false false

Set current 580 580 580 770

RMS current 994 550 550 764

MAX current 1402 776 776 1077

Run current 17/31 17/31 17/31 24/31

Hold current 8/31 8/31 8/31 12/31

CS actual 0/31 8/31 8/31 12/31

PWM scale

vsense 0=.325 1=.18 1=.18 1=.18

stealthChop false true true true

msteps 256 16 16 16

tstep 0 max max max

PWM thresh.

[mm/s]

OT prewarn false false false false

pwm scale sum 0 10 10 14

pwm scale auto 0 0 0 0

pwm offset auto 0 36 36 36

pwm grad auto 0 14 14 14

off time 0 4 4 4

blank time 16 24 24 24

hysteresis

-end -3 2 2 2

-start 1 1 1 1

Stallguard thrs 0 0 0 0

uStep count 0 40 40 56

DRVSTATUS X Y Z E

sg_result 0 0 0 0

stst *

olb

ola

s2gb

s2ga

otpw

ot

157C

150C

143C

120C

s2vsa

s2vsb

Driver registers:

	`X	0x00:00:00:00	 Bad response!`

	`Y	0xC0:08:00:00`

	`Z	0xC0:08:00:00`

	`E	0xC0:0C:00:00`

Testing X connection... Error: All LOW

Testing Y connection... OK

Testing Z connection... OK

Testing E connection... OK
```

And as you can see, the values:

```
vsense 0=.325 1=.18 1=.18 1=.18

stealthChop false true true true

msteps 256 16 16 16

tstep 0 max max max
```

Are showing that the X-Axe have really different values and at the bottom it said:
```
Driver registers:

	`X	0x00:00:00:00	 Bad response!`

	`Y	0xC0:08:00:00`

	`Z	0xC0:08:00:00`

	`E	0xC0:0C:00:00`

Testing X connection... Error: All LOW

Testing Y connection... OK

Testing Z connection... OK

Testing E connection... OK
```

Bevor I tried to made the upgrade every Axe was working fine, so it can't be hardware issue.
There is a way to solve this?

basurci, 2020-12-06:
If you purchased board with TMC drivers that have UART already enabled than you won't get this problem.
Double check your stepper drivers if they have soldering or not for UART mode.
Otherwise you can do it yourself.
[image]

rcxrbt, 2020-12-31:
By chance, you connect the USB cable before turning on the printer?