bigtreetech/BIGTREETECH-SKR-mini-E3 #465: sd-card not reading SKR-mini e3 V2.0

briandewin, 2020-12-04:
After I attached media my sd card got corrupted after that no sd-card is being read, the board itself has the green indicator turn green when a card is inserted but it just wont read it, is there anything I can do to fix this?

basurci, 2020-12-06:
Does your board connect to PC. If it does you can flash directly using Visual studio.
If there is no connection to PC try another SD card, if that one also doesn't work you may have corrupted STM32 chip.
If you tried to flash Klipper on board than you bricked your board. Klipper only works on 1.3 version.

briandewin, 2020-12-06:
I can flash trough pc but that does not solve the issue, another sd card is also not recognisable, i did not try to flash klipper, it was conected with octoprint on my raspberry pi and when i deleted a file on my sd card in octoprint the sd card got corrupted now nothing is recognised in the sd card slot, i tried putting a stock firmware on it but i cant seem to flash it trough visual studio does anyone have a file so i can install it with platformio?

basurci, 2020-12-06:
Is your board being recognized by octoprint? Or it can't connect to it?

thebiggestgit, 2020-12-09:
There is a firmware update for the Bigtree/SKR TFT screens which addresses this, assuming you were using that as well? Just flash that with the latest firmware if so and it should sort out if you are using one.

winchester7mm3d, 2020-12-11:
I have almost the same problem, brand new SKR mini e3 v2.0, and it can only read the SD card that came with the board.  I tried reformating another card in FAT32 and NTFS and both times...nothing.  Any ideas?

briandewin, 2020-12-11:
I fixed the problem by installing a new firmware but with usb and visual code, now i can install new firmwares and i can print again with my sd card

briandewin, 2020-12-11:
> I have almost the same problem, brand new SKR mini e3 v2.0, and it can only read the SD card that came with the board. I tried reformating another card in FAT32 and NTFS and both times...nothing. Any ideas?

Maybe a firmware reinstall might fix it for you to?

winchester7mm3d, 2020-12-11:
Briandewin,
Thanks, I will have to try that.  I just reinstalled the most recent .bin  (I copied it onto the only SD card it would read), tried again with the other card and no luck.

I will try this weekend via USB and see what happens. Thanks again!!

markbwalker, 2021-01-08:
Same problem here, and I even had a flexible SD micro card extender to eliminate the constant in/out of the SD card.

I did notice the SD card is recognized if I push it in a little further (but if I let loose, then it ejects).  Green light went on even when a card wasn't recognized...so that doesn't mean much apparently.

bartosslajos, 2021-01-10:
Same problem here, I had to push the MicroSD card during the complete Boot/Flash process, and then the Flash was succesfully! So it is some stupid mechanical issue...

markbwalker, 2021-01-10:
> Same problem here, I had to push the MicroSD card during the complete Boot/Flash process, and then the Flash was succesfully! So it is some stupid mechanical issue...

The board is almost new (1 month), so at some point a replacement should be headed my way.  But I'm tempted to get the BigTree display in addition since it has a slot for a memory card.  I won't really trust the SD microcard connector going further since I've seen other comments regarding the issue (hence the original use of the extender cable for the microcard).

bartosslajos, 2021-01-11:
> > Same problem here, I had to push the MicroSD card during the complete Boot/Flash process, and then the Flash was succesfully! So it is some stupid mechanical issue...
> 
> The board is almost new (1 month), so at some point a replacement should be headed my way. But I'm tempted to get the BigTree display in addition since it has a slot for a memory card. I won't really trust the SD microcard connector going forward since I've seen other comments regarding the issue (hence the original use of the extender cable for the microcard).

I use Octoprint + RaPi, so I don't need a microsd connector at daily use, so I'm lucky I guess... :)

markbwalker, 2021-01-13:
I reached out to BigTreeTech about the defective SD connector. After sending two videos, they asked if I could replace the connect myself or have it repaired (both with reimbursement).  I offered what I thought was a simple compromise - if they could send a TFT display I could just use that memory card slot instead, meaning I don't have to repair, they don't have to fool around with reimbursement - just cheaper all the way around and the easiest.  Not sure why they just didn't offer a new board in the first place, given the cheap price of the motherboard.   Anyways, the next reply was a $10 refund offer.  Clueless why they think since I have a broken motherboard, send me $10 and all is good.  You would think making sure the customer has a working product might be a better option.   As of right now, I obviously declined and will just return to Amazon.  Even if I would get a replacement, I don't trust the connector, and customer service resolution on a brand new board is even less.  So I'll look into the Creality 4.2.7 board or other options instead instead.

philipszalla, 2021-07-11:
Hi @briandewin,

I have a SKR mini E3 V2.0 and it can't read SD cards, too.

How did you update your firmware using vscode and usb?

bartosslajos, 2021-07-11:
During the flash process you need to hold/push the sd card in the slot.

Philip Szalla ***@***.***> ezt írta (időpont: 2021. júl. 11.,
V 13:45):

philipszalla, 2021-07-11:
Hi @bartosslajos,

I tried to push the sd card in the slot. But no success.
I tried with 3 different sd cards (128MB provided by skr min, 8GB provided by Ender 3 Pro, 32GB from SanDisk).

Any idea?

coxml32, 2021-07-11:
You should be inserting the SD card with the copper runs facing up.

philipszalla, 2021-07-11:
Hi @coxml32,

yes, I know.

I own the board since 11-2020. The SD card slot works fine at the begining. But since a few months it does not recognize the sd cards. Before it was no problem because I use octoprint. But now I need to update the firmware. And that seems not possible without working SD slot :/