bigtreetech/BIGTREETECH-SKR-mini-E3 #467: Runout

TimRexuino, 2020-12-08:
I have tried everything I can think of on the firmware but the runout sensor doesn't pause or stop the printer.

t0b4cc0, 2020-12-11:
but how can we know what you did think of to try?

creatron, 2021-01-02:
M119 display the status.  Send command and verify that open/trig ired happens when filament is removed/ inserted. I change pins in pins_BTT_SKR_MINI_E3_common.h from PC15 to PC12 because I use the E0 end stop and it works for me. Marlin had a bug that you could start a print without filament, but it stops when filament detector run-out.  Not sure if 2.07 had this fixed.
//
// Filament Runout Sensor
//
#ifndef FIL_RUNOUT_PIN
  #define FIL_RUNOUT_PIN                    PC12   // PC15  // "E0-STOP"
#endif