bigtreetech/BIGTREETECH-SKR-mini-E3 #468: [BUG] Cura brings BTT SKR mini E3 V2 to reboot with alarm before when opening

Bastlwastl84, 2020-12-09:
### Description

Starting Cura crashes Printer
### Steps to reproduce

1. [First Step]
Connected with USB
2. [Second Step]
open Cura 4.8
3. [and so on...]
rebootloop till unconnect or closing Cura. 

**Actual behavior**

Reboots when opening Cura or plugging USB cable in when Cura is open

### Additional Information
BTT TFT35 E3 V3 2020-12-05
Marlin Bugfix 2.0.8 2020-12-05
[2020-12-09.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5662825/2020-12-09.zip)

delamonpansie, 2020-12-22:
Workaround: edit `.config/cura/plugins.json` and add `"USBPrinting"` to `"disabled"`.
Example: `{"disabled": ["USBPrinting"], "to_remove": [], "to_install": {}}`

Bastlwastl84, 2020-12-22:
Couldn't there be no workaround in Marlin creating? Everytime after Cura updating it must be fixed.

mkotyk, 2022-01-13:
@delamonpansie Not really a work around, since you can no longer print to the printer.

It's 2 years later and I am seeing this also with Cura 4.13 and Marlin from the BTT github repo.  I guess no one cared.

mkotyk, 2022-01-13:
In case anyone else ends up here, the real fix is https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/499