bigtreetech/BIGTREETECH-SKR-mini-E3 #471: [BUG] (short description)

Stormwing01, 2020-12-12:
### Description

Tmc connection error. I just turned on my printer today after it's been working perfectly for the past 2 weeks, and i suddenly get bombarded with this error.
all the steppers still work, and pronterface says all my drivers are "ok"

### Steps to reproduce

apparently just use the btt skr mini v2 for 2 weeks.

**Expected behavior**
no errors.

**Actual behavior**
errors.

### Additional Information
in marlin mode it says "tmc connection error" and in tft mode it says "all low"
it also refuses to start up in marlin mode when I start the printer.

Stormwing01, 2020-12-15:
Edit: I have disabled eeprom_settings and the error has disappeared. but this also means I cannot save settings, so I cannot keep it disabled.

any idea of a fix?

rcxrbt, 2020-12-31:
By chance, you connect the USB cable before turning on the printer?

Stormwing01, 2020-12-31:
> 
> 
> By chance, you connect the USB cable before turning on the printer?

Usb was never involved. I updated the motherboard directly from sd card. (its a btt skr mini v2)

as of right now the error is gone.... unless I set my y stepper driver current below 450mA and it disappears when I set it anywhere above that current... but even when I get that error everything works perfectly and both the terminal on the tft35 I have as well as pronterface says my drivers are ok.