bigtreetech/BIGTREETECH-SKR-mini-E3 #476: Connecting PT100 to SKR mini E3 v2.0 board

dovcharov, 2020-12-24:
Good day, everyone.

I have a PT100 temperature sensor with amplifier board, which I have ordered from E3D-online (https://e3d-online.com/collections/upgrades-v6/products/v6-pt100-upgrade-kit). Now I plan to upgrade my CR-10 3D printer to SKR mini E3, and I want to install this PT100 into my hotend instead of Semitec 104GT sensor.

Can someone please recommend, what is the best way to connect PT100 to my SKR mini E3 2.0 board?

I have thoroughly researched this topic, and read some posts about PT100 installation here
https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/issues/179
and here (in russian)
https://mehanik.me/ru/blog/bigtree_tech_skr1_4_tmc2209_bltouch_pt100_tft35_v2/

As far as I could understand, there are several installation options for PT100. Please correct me if I'm wrong!
1. Install PT100 and connect it to hotend thermistor pins directly (option 147 in Marlin bugfix 2.0.x). Simple, but very inaccurate temp readings.

2. Install PT100 and connect it to hotend thermistor pins directly, desolder 4.7K pullup resistor and replace with 1K pullup resistor. Requires solder skills, less inaccurate temp readings. Option 110 in Marlin bugfix 2.0.x.

3. Install PT100 via E3D adapter board. Adapter board needs to be powered via 5V, signal pin needs to be connected to ADC input on the SKR board. The ADC signal output on adapter board needs to be protected by Zener diode (stabilitron) so that voltages above 3.6 volts don't go to the printer board's input (SKR mini E3 board CPU uses 3.3 volt logic). Option 21 in Marlin bugfix 2.0.x. 
This is one of the optimal connection methods as far as I understood, with accurate temp readings. But I have failed to locate the spare ADC In pin on the SKR mini E3 printer board. As I understood, hotend thermistor pin can be set up as ADC IN, but then it's affected by the voltage passing through pullup resistor on hotend thermistor pin. So 4.7K pullup resistor needs to be de-soldered. Complexity of such connection is relatively the same as for the option 2. 
Is there a spare ADC IN pin on the SKR mini E3 2.0 board, which can be used?

4. Install PT100 via E3D adapter board. Adapter board needs to be powered via 3.3V, signal pin needs to be connected to ADC input on the SKR board. The ADC signal output on adapter board doesn't need protection in such case, but there's a small chance that amplifier board won't work, because it was originally designed for 5V input. Option 20 in Marlin bugfix 2.0.x. 
This is the recommended connection option with the most accurate temp readings. But the challenge for such connection option is the same as for option 3. It needs a spare ADC_In input, or I need to de-solder 4.7K pullup resistor on hotend thermistor pin.

5. Install PT100 via MAX31855 SPI adapter board. Such connection method is covered in "[GUIDE] to using PT100 or PT1000 or Thermocouples with SKR PRO V1.x boards or GTR V1.0 boards" linked above. But this connection method requires a CS (chip select) pin to be assigned for MAX31855 board, so that SKR board CPU could select SPI adapter board during poll cycle. 
Which pin can be used as CS pin on SKR mini E3 v2.0 board?

Thanks in advance for your advices.

dovcharov, 2020-12-25:
Just understood that I can use TX2, RX2 pins on SKR mini E3 board (in case I don't use TFT screen), or Servos (PA1) pin as ADC-IN. 

https://yandex.ru/images/pad/search?text=stm32rct6%20pinout&img_url=https%3A%2F%2Fwww.mikrocontroller.net%2Fattachment%2F50626%2FSTM32F103xx_LQFP64_pinout.PNG&pos=0&rpt=simage&stype=image&lr=146&source=wiz
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0SCHpdf.PDF
SKR

GadgetAngel, 2020-12-25:
@dovcharov I hope this helps:

[image]

dovcharov, 2020-12-26:
This. Is. It! My sincere gratitude to you, kind sir!

no1jester, 2021-01-05:
@dovcharov how did you end up wireing yours up?
I Have the PT100 and Amlifier Board from e3d with the BTT SKR Mini E3 but having a little trouble getting it to work even after studying the above want confirmation as to weather im wrong or my amp boards faulty

dovcharov, 2021-01-05:
Not yet. My SKR mini E3 board is still being shipped to me. I will get it in 2 weeks. I will reply here once I try PT100 with it.
I also ordered a Max31865 spi board, will connect PT100 through it if analog amplifier board doesn't work.

GadgetAngel, 2021-01-05:
@no1jester How are you trying to hook up the E3D PT100 amplifier board to the SKR mini E3 v2 board? Which analog pin on the SKR min E3 are you trying to use?

GadgetAngel, 2021-01-05:
Here are the options you have on the SKR MINI E2 V2.0 board:

[image]
 

You hook up the PT100 via **Analog Procedure** or via **Digital Procedure**.  The **MAX31865 board** is used **only** for the **Digital Procedures**.  The **E3D PT100 Amplifier board** is used for the **ANALOG PROCEDURE**.  These are two totally different ways of connecting up the PT100 Temperature sensor.  

To use the Analog Procedure you MUST find a PIN on the SKR MINI E3 V2.0 board that is an **A/D INPUT PIN** and is not being used by the board for something you will need to make your printer work.   By looking at the schematic in combination with the stm32f103rc.pdf file which is the SKR MINI E3 V2.0 processor's data sheet. I see that you have some options.  But if you do not want to hack at the SKR MINI E3 V2.0 board you only have TWO options for the Analog Procedure to hook up the E3D's PT100 amplifier board: PIN PA2 and PIN PA3 which are TX2 and RX2 on the TFT connector.  If you want to use the E3D's PT100 board instead of the MAX31865 board you will have to use a CR10_STOCKDISPLAY and hook it to EXP1 connector or do not use a display at all and use Octoprint via the USB connection.  But you will not be able to use a TFT screen because TX2 and RX2 are analog PINS and they are the only PINS that are not being used by the printer that are ANALOG PINS.

Here is the stm32f103rc.pdf file: 
[stm32f103rc.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5772047/stm32f103rc.pdf)

Here is the schematic for the SKR MINI E3 V2.0 board: 
[BTT SKR MINI E3 V2.0SCHpdf.pdf](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5772057/BTT.SKR.MINI.E3.V2.0SCHpdf.pdf)

GadgetAngel, 2021-01-05:
@no1jester Here are two wiring diagram that I hope will help:

User **Thermistor table number 20**:
[image]

Use **Thermistor table number 21**:
[image]

no1jester, 2021-01-06:
Yeah I think I might do the same I can't seem to get the e3d pt100 to read
correctly

GadgetAngel, 2021-01-06:
@no1jester Check your PT100 sensor with an Ohm meter. It should read around 110 Ohms at room temperature.

no1jester, 2021-01-06:
Yeah it's at 108 I am leaning towards board 170 room temp is Def not right
haha

dovcharov, 2021-01-22:
@GadgetAngel, I have a question. If I use MAX31865 board, there are 2 modes available - hardware SPI and Software SPI. Which of two is better and why?

IdeYoda, 2021-02-27:
Good day for everyone,
i was getting here by seeking for a similar solution, maybe you can help me too. I just installed a RPI4b beside my Ender3 Pro and would like to leave off the USB cable between the two by making connection directly in the house. I find a very nice solution (see the link below) but for a different motherboard and i'm not sure which would be the useful send and receive pin for me on BTT SKR mini V2. It would be also very useful as it doesn't need any power supply just data in-out and a ground. To show better what i mean here is the same project on a different board:
[https://www.tardigradeoutdoors.com/blog/how-to-print-without-a-usb-cable-on-an-ender-3-and-octoprint](https://www.tardigradeoutdoors.com/blog/how-to-print-without-a-usb-cable-on-an-ender-3-and-octoprint)

GadgetAngel, 2021-02-27:
> @GadgetAngel, I have a question. If I use MAX31865 board, there are 2 modes available - hardware SPI and Software SPI. Which of two is better and why?
> I decided to connect PT100 with Max31865 board, and use PC14 pin as CS. I can use Z-Endstop connector for BLTouch instead of PC14 pin.

The best solution in my opinion is the choice that allows you to use the MAX31865 board. If both hardware SPI and Software SPI are available then I would use Software SPI because the Hardware SPI implementation for the SKR E3 V2.0 board shares the SPI lines with the on board mini-SD card reader. Anytime you share lines with another device there is always a chance that the two peripherals could interfere with each other.

GadgetAngel, 2021-02-27:
> Good day for everyone,
> i was getting here by seeking for a similar solution, maybe you can help me too. I just installed a RPI4b beside my Ender3 Pro and would like to leave off the USB cable [...]

The way you connect up the RPI4B to the SKR E3 V2.0 board is to see which UART ports are available to use for the board.  If you look at the SKR E3 V2.0 schematic located:https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0SCHpdf.PDF

you will see that their are two USART ports available TX1,RX1 and TX2,RX2.  Therefore for the Bigtreetech SKR E3 V2.0 board the following numbers can be used in the configuration.h file for the following Marlin Variables: SERIAL_PORT and SERIAL_PORT_2.  The following values can be used: -1 for USB Port; 1 for EXP1 connector; 2 for TFT connector.

First you have to decide which connector is free or available to use: EXP1 connector or TFT connector.  If EXP1 connector is available to use then use TX1,RX1.  If TFT connector is available to use then use TX2,RX2.  If both are available make a choice.  

I will assume you are using EXP1 connector for your display screen.  Therefore the TFT connector is free or available to use to hook up the RPI4B.  Connect TX2 from the SKR E3 V2.0 to RX on the Raspberry Pi 4B.  Connect the RX2 from the SKR E3 V2.0 board to the TX on the Raspberry Pi 4B.  Connect the GND from SKR E3 V2.0 board to GND on the Raspberry Pi 4B.

In Marlin's configuration.h file set SERIAL_PORT_2 to -1 and set SERIAL_PORT to 2.

To setup the raspberry pi 4B, follow the instructions found in the SKR PRO V1.1 manual that I wrote.  See appendix I, appendix J or appendix K, but use the SERIAL_PORT_2 as -1 and SERIAL_PORT as 2. https://github.com/GadgetAngel/SKR-PRO-V1.1-Stepper-Driver-Jumper-Configuration-Manual/blob/master/CURRENT-Manual/SKR%20PRO%20V1.1%20Stepper%20Driver%20Jumper%20Configuration%20Version%206.7_reducedFileSize.pdf

I hope this helps

IdeYoda, 2021-02-28:
**@GadgetAngel**  Yes, it helps a lot, i'm grateful. I'm not into those things a lot - we can say an idiot in electricity -  but this sounds easy enough to follow, thanks a lot! :)

IdeYoda, 2021-03-03:
> To setup the raspberry pi 4B, follow the instructions found in the SKR PRO V1.1 manual that I wrote. See appendix I, appendix J or appendix K, but use the SERIAL_PORT_2 as -1 and SERIAL_PORT as 2. [...]

Dear @GadgetAngel,
after all it turned out both of my motherboard RX/TX connections are in use as i have a BTT TFT35-E3 V3.0 monitor too which is useing both. But I also discovered the TFT panel has a not only one but immediatlly two UART ports with the necessary connections. Even further, it also has a 5V power output which made me wondering if i could power up the RPI in the house with some soldering on RPI board and forget the RPI power cable also. I'm I right? Is this panel wireing capable to support the necessary 2A too?
[https://github.com/bigtreetech/BTT-TFT35-E3-V3.0/blob/master/Hardware/BTT%20TFT35-E3%20V3.0PIN.pdf](url)

GadgetAngel, 2021-03-03:
> Dear @GadgetAngel,
> after all it turned out both of my motherboard RX/TX connections are in use as i have a BTT TFT35-E3 V3.0 monitor too which is useing both. But I also discovered the TFT panel has [...]

First you should not power the RPI from the 5V source on the SKR E3 V2.0 board or from the back of the TFT screen. Use an independent power supply for the raspberry pi. 

Second the TFT screen you are using does not require the use of both UART connections. If you disconnect the TFT part you lose the touch screen. If you disconnect the EXP1 then you lose the legacy screen. You can not use the UART connections on the back of the TFT screen to connect up the raspberry pi. The purpose of connecting up the raspberry pi via UART is to communicate with the BIGTREETECH BOARD directly. The TFT screen does not talk to the BIGTREETECH board directly. If you connect the RPI to the TFT screen UART ports you will be talking to the MCU on the back of the TFT screen. That is why you need to connect up the RPI to the UART ports on the SKR E3 V2.0 board. If you connect up the RPI to the UART ports on the TFT screen you will communicate with the MCU on the TFT screen. If you connect up the RPI to the UART ports on the SKR E3 V2.0 board then you will be talking to the MCU on the SKR E3 V2.0 board

IdeYoda, 2021-03-03:
Thanks, ok, that was my last dumb question... Thanks again!

no1jester, 2021-08-20:
Hey guys how is everyone going with this i am struggling with configuring Marlin 2.0 does anyone have a config i can download?

GadgetAngel, 2021-08-21:
> Hey guys how is everyone going with this i am struggling with configuring Marlin 2.0 does anyone have a config i can download?
> 
> Ok i managed to compile it not really sure what happened i just closed VS re opened hit build and it worked now to test it on the machine :)

I have experience that myself.  If I have an issue I always shut down VScode  to restart it then try and recompile.  For some reason if you make a change to board type being used VScode can get confused.  Sorry you had a problem but I am glad you tried the restart and that restarting VScode solved your problem.  

Have a great day.

no1jester, 2021-08-22:
> I have experience that myself. If I have an issue I always shut down VScode to restart it then try and recompile. For some reason if you make a change to board type being used VScode can get [...]

Thanks for all your help here my last issue is i am getting EEPROM datasize error. All Low and i am showing a -15 on the nozzle temp so not sure if its an issue with the MAX board ill have to check that because i reverted back to straight connection and it works fine i might rever to 1.9 marlin i dont think my ender likes it allthough its not really an ander anymore haha i have a BTT board with a E3D Titan Aqua hot end :)

jantman, 2023-04-10:
Absolutely amazing information in this thread, thanks so much! @GadgetAngel is there any chance that you have similar wiring diagrams for the SKR Mini E3 V3 (as opposed to V2)?