bigtreetech/BIGTREETECH-SKR-mini-E3 #479: [BUG] skr mini e3 v2 manual bedleveling values always +200

gpelekanos, 2021-01-01:
### Description

 skr mini e3 v2 manual bedleveling values always +200

### Steps to reproduce

1. Betleveling
2. Store Settings 
3. Prinding 
4. Betleveling 

**Expected behavior**
Bedleveling values saved 

**Actual behavior**
Z Values are not saved. The new leveling begint with all Z values set bei +200. Bei prinding The nozzle is far away up from the bed.

sarvenn, 2021-01-10:
I have the same issue.
I also use TFT35 e3 v3. Z offset menu warns me M851 unknown command. So I can not set the Z offset before printing. I do the manuel mesh bed leveling.
While printing I do babystep Z and store the value. However I can not see that value in M503 G29 bed z line. After printing and starting to a new print I open the babystep menu and it show 0 even I adjusted it to a different value at previous print.

ledicjp, 2021-01-23:
I have the same problem, babystep values are not saved.
Ender 3, skt 2.0, bltouch 3.1, tft 3.1

sarvenn, 2021-01-23:
#1429

sarvenn, 2021-01-23:
Please check this topic and try the latest tft fw.

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1429