bigtreetech/BIGTREETECH-SKR-mini-E3 #480: [BUG] firmware-bltouch-for-z-homing.bin - does not allow for negative z Offset

nuno1895, 2021-01-02:
### Description

### Steps to reproduce

1. [First Step]
2. [Second Step]
3. [and so on...]

**Expected behavior**

Expect to be able to offset z

**Actual behavior**

Will not allow z-offset to go past zero

### Additional Information

Dont know how to do any of this

* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

razaci, 2021-01-03:
Exact same problem. waiting for result.

sorerethue, 2021-01-03:
same issue found. Not able to home Z or move z axis towards the bed

razaci, 2021-01-03:
i had to install vb and rebuild my configuration files then my printer is running fine.

Chillystout, 2021-01-09:
It's a safety feature. Use the command M211 S0 to switch off soft endstops, then you can have negative values. After finishing your procedures, to switch on the soft endstop use the command M211 S1.
To insert the command you can use Cura and the printer attached to the PC/laptop.