bigtreetech/BIGTREETECH-SKR-mini-E3 #481: SKR Mini E3 V2.0 BLTouch isn't working, only Blue LED, no Red LED, no self test

narmer6, 2021-01-03:
Just replaced a SKR Mini e3 V1.2 with a SKR Mini e3 v2.0, now BLTouch isn't working.

If I flip the Z-stop pins, black wire top and white bottom, the BLtouch lights up solid blue LED, no self test, no deploy/stow, no red LED

I have tried rearranging the pins on Z-stop, and rearranging pins on other 3 cable connector. If I plug in with correct configuration I get no LED, no blue, no red, if I plug in Z-stop cables flipped I get solid blue LED.

I have tried using firmware that works with a different printer using SKR Mini e3 v2.0, I have tried flashing newest firmware, and I have tried using preconfigured firmware from BTT github

Is there a problem with the board? Is there a problem with the BLTouch? The BLTouch was working before I changed the boards.

I have tried using a BIgtreetech BLTouch cable extension, and the stock antclabs BLTouch cable, no difference

peterjmetcalfe, 2021-01-04:
I have the same issue and done everything you have done to troubleshoot as well. I have not yet figured out the solution, you are not alone. Not sure if its the SKR board, I actually replaced the probe as an additional troubleshooting attempt and am having the same issue. So I do not think its the probe's hardware, though I'm not sure. I suspect it is user error, I cannot find any video guides or posts talking about this issue.

dilworthkc, 2021-01-06:
almost have the same problem, skr mini 1.2 bltouch v3.0  pluged into probe, keeping the oem z stop connected,  when i turn on the printer it does self test. but will not function beyond that, flashed the bltouch firmware from bigtree, orginaly had it hooked up as the z stop before i switched it back because it sometimes wouldnt work and was unreliable. pluged in octoprint, and now my bltouch is just a clicking reminder that i turned on the machine.

ellensp, 2021-01-08:
No deploy/stow shows there is an issue with the 3 wire servo part of the bltouch. 
There are two option to plug in the the servo cable.
1) you plug it into Z probe plug. Check the blue or black wire is on the GND pin, the red wire in on 5v and the yellow in on PA1
This is default for the firmware provided by BTT
2) you may have one of these "pin 27 boards" that allows you to use the beeper pin on the LCD as the servo pin.
The firmware is not setup for this
You would need to update "#define SERVO0_PIN"  in  pins_BTT_SKR_MINI_E3_common.h with the correct pin from the LCD connector.  looks to be PB5

coconitro, 2021-01-09:
Running into the exact same issue as the OP, same diagnostic steps. Wondering if this is related: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/484

YBeyin, 2021-01-12:
Did you change the order of the pins on the dupont? You must the change the order of pins if you are using original Bl-Touch

They must be like this. Just very slightly lift the marked place and pull the cable then put them back as like this picture: 
[image]
Also connect them as like this on the mainboard:
[image]

egecko-cmd, 2021-01-16:
For some reason, I was having issues with compiling with VScode. I gave up for the time being and will temporarily live with the E3 Pro having a Z 250mm, instead of the current setup of Z 350mm. I just used the BTT supplied firmware and changed the needed configurations using the gcodes.

-Printed a BL mount
-Installed BL on z-probe connector only and DID NOT install BLtouch for Z-Stop (wiring diagram and info below).
-Download and made the appropriate name change to the file firmware-bltouch.bin
-I measured and changed the BLtouch offsets (because using a printed mount its required, but using the metal mount doesn't require changes).

Steps for those needing help.
1. First off, update to correct firmware v2.0 and decide of you want firmware-bltouch.bin or firmware-bltouch-for-z-homing.bin
2. Of course, MUST place the file on the SD's main directory and change the name to "firmware.bin" so it updates. 
3. Perform bed leveling and update all calibration parameters as needed.

Swapping the pins are not needed if you're using the latest versions of BLtouch (current is v3.1).

This is here for those that may become confused and assume they need the DCDC module 
The install manual (2/3 down) located in the hardware section, can be confusing as it states:

**IV Interface Instruction  
#4 5V power selection for BLtouch, TFT and RGB interfaces:
1. On Board +5V power supply (jumper move)
2. Use Bigtreetech DCDC v1.0 module +5V power supply.** 

but........the PIN diagram states "NEO-PWR" on the board for the jumper. The DCDC module is needed if you're installing LEDs.

It makes it read as if you must add the DCDC module for BLtouch, TFT and RGB interfaces and move the jumper. You don't need to move the jumper unless you're using up to 30 LEDs. (side note: the previous version of DCDC module WILL NOT FIT on V2.0 board).

#5 Connection with BTT relay v1.2: (shows the pin diagram of the Z-probe)
(You don't have to use the Z stop function and can just connect all 5 wires to the Z-probe connector as below.) 

Pin/wiring diagrams and placement
 
Z-Probe connector      BLtouch Wires
[] PC14                        White (Zmin)
o GND                         Black (GND)
o PA1                          Orange (control signal)
o +5V                          Red (+5V)
o GND                         Brown (GND)

BLTouch Wiring from v3.1 Manual. Link below has the all versions available.

3Pin: Brown (GND), Red (+5V)Orange (control signal); 2Pin: Black(GND) White (Zmin)

Links:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0%20Instruction%20Manual.pdf

https://www.antclabs.com/manual?lang=ko

excuse if a lot of information. it's late and I'm tired.

MarkyPocock, 2021-01-31:
Also having similar problems with my BLTouch, except its stuck on red and did occasionally work but now seems to be consistently not working. 

Turn power on, BLTouch lights up red and deploys then retracts the probe a couple of times as per normal but never goes blue. Then if I try and home it never deploys and just crashes into the bed despite it working previously (and yes I have the right firmware for z homing as I said it worked sometimes previously). None of the test buttons respond: https://youtu.be/rGj2VmXeh3I

I can only think its firmware related but not sure why its gone suddenly from sometimes working to never working. I will opening it and make sure nothing has come loose later but it does feel like a firmware issue.