bigtreetech/BIGTREETECH-SKR-mini-E3 #482: Firmware 2.0.7.2 preconfigured never stop the fan during printing. To print ABS i need to stop fan.

cybsit, 2021-01-04:
Hi, I'm using BIGTREETECH SKR mini E3 V2 on an Ender 3 Pro printer with BlTouch.  I loaded the latest preconfigured firmware with BlTouch and found that I can't print ABS because the fan never stops during printing even if the fan speed is set to 0% or no cooling.

Seelenkind, 2021-01-04:
The chip for the fan breaks very quickly. To test, you can downgrade to Marlin Bugfix 2.0 without BL Touch. If it still runs continuously, the board is defective and has nothing to do with the firmware.

cybsit, 2021-01-05:
Hi, the fan runs all the time when it prints even if I set it without a fan.  when it finishes printing, the fan stops.

davidrichter-cz, 2021-01-22:
Hi, I have the same problem with the blower fan, I do not have a BL touch, the problem occurred by changing the default display to TFT24. So I don't believe there is a bug on the E3 MINI board but in the firmware. Any advice?

cybsit, 2021-01-22:
when I want to print ABS or PETG I use firmware 2.0.6.1 which does not start the fan at all.  for other prints where it doesn't matter that the fan is running all the time I use firmware 2.0.7.2

cybsit, 2021-01-22:
hopefully these issues will not appear in the next firmware update

alvarini, 2021-02-04:
I have compiled my own Marlin with last version available and when plugging layer Fan to Fan 0 is always 100% and not responding to pwm signals

davidrichter-cz, 2021-02-04:
Problem solved, just flip the fan connectors, then everything works as it should.

davidrichter-cz, 2021-02-04:
Problem solved, just swap the connectors FAN0 and FAN1 fans, then everything works as it should.

cybsit, 2021-02-04:
it worked for me, thanks.  case closed!!!

closed 2021-02-04 by cybsit

alvarini, 2021-02-05:
not solved for me, I flipped the connectors and I got fire in the black component over the Fan0 connector

davidrichter-cz, 2021-02-05:
Swapping connectors FAN0 and FAN1 cannot cause a fire. You have another fatal mistake somewhere. Pay attention to 12V and 24V power supply and fan voltage. Pay attention to the polarity of the FAN!

> nevyřešený pro mě, otočil jsem konektory a dostal jsem oheň v černé komponentě přes konektor Fan0