bigtreetech/BIGTREETECH-SKR-mini-E3 #484: Skr e3 mini V2 Not getting 5v on the 5v zprobe pin

clementjr87, 2021-01-08:
im trying to install a bl touch 3.1 and its not working at all i have defined 5v in the firmware and im not getting 5v on the pin on the zprobe

coconitro, 2021-01-09:
Same issue (was the root cause of this: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/481#issuecomment-757085901)

Niether  BLTOUCH_FORCE_MODE_SET nor BLTOUCH_SET_5V_MODE and both jumper positions get the Z-PROBE 5V to work.

clementjr87, 2021-01-09:
I have the jumper in the original position it came in on the board. I have both BLTOUCH_FORCE_MODE_SET and  BLTOUCH_SET_5V_MODE defined .I'm in contact with support and I sent them a video. I'm not getting v5 on the jumper either. I think support is handling in though. ThanksSent from my Galaxy

Same issue (was the root cause of this: #481 (comment))

radek8, 2021-01-11:
Do you have the jumper in the correct position?
[image]

coconitro, 2021-01-13:
Neither position has worked for me.

abacabie, 2021-01-13:
my board arrived with a dead Z-endstop, I never got 5v with it, I had to define the Z-endstop to PWR-DET in the pins file of marlin.

radek8, 2021-01-13:
> moje deska dorazila s mrtvým Z-endstopem, nikdy jsem s ním nedostal 5V, musel jsem definovat Z-endstop na PWR-DET v souboru pinů Marlin.

Was the Dead data pin or a 5V power pin?

radek8, 2021-01-13:
The board has a 32b processor with 3.3V logic.
The data pin is connected to 3.3V via a 10k ohm resistor
So there must never be 5V on it. Theoretically, a 5V input can destroy it.

[image]

egecko-cmd, 2021-01-16:
> Do you have the jumper in the correct position?
> [image]

The jumper location is marked as NEO-PWR in the PIN diagrams.

The default jumper location is for up to 8 LEDs and only needs to be moved if using up to 30 LEDs.