bigtreetech/BIGTREETECH-SKR-mini-E3 #485: [FR] Add PID bed autotune and TMC debug

rcxrbt, 2021-01-10:
Hi team,
 can you enable by default PID bed autotune and TMC debug on pre-compiled firmware?

radek8, 2021-01-11:
It is always better to compile your own current FW.
Not this old one

rcxrbt, 2021-01-12:
Sure, but why not make it as default?

radek8, 2021-01-12:
If you consider it important, you can create Pull Requests.
I would never download FW from here. It is obsolete.
Use instead
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

rcxrbt, 2021-01-12:
Of course, but I'm referring to the pre-compiled firmware that we can find here.

egecko-cmd, 2021-01-16:
> If you consider it important, you can create Pull Requests.
> I would never download FW from here. It is obsolete.
> Use instead
> https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

of course, but when in a pinch......

I had issues compiling FW and just put the bltouch on my v2.0 as I will deal with it later.

Seelenkind, 2021-02-03:
TMC_DEBUG is no longer needed in Marlin 2.0.x, but enabling it produces an extended report.
Responses of all LOW (00:00:00:00) or all HIGH (FF:FF:FF:FF) are signs of a communication problem.
See TMC Troubleshooting for further information.
https://marlinfw.org/docs/gcode/M122.html

You need preconfigured firmware with
Temperature PID Tuning , Printer statistik, resume after power loss, mesh bed leveling?
look at this 
https://github.com/Seelenkind/BIGTREETECH-SKR-mini-E3/tree/Marlin-2.0.7.2-SKR-mini-E3-V2.0/firmware/V2.0

closed 2021-06-15 by rcxrbt