bigtreetech/BIGTREETECH-SKR-mini-E3 #487: Blank blue screen, board seems to be DOA

g2bb, 2021-01-12:
I installed my SKR mini E3 into my CR-10 yesterday being careful to plug everything back in where it should be. When I turned it on I expected to have a working system, but the screen was completely blank. I've tried everything I can find on the internet short of buying a new board. I have reflashed the firmware (BTT official and individual users firmware) making sure it was 200+ kb, plugged in only power and display, reset the board. When flashing the firmware I don't see any flashing lights, but after, the firmware.bin becomes firmware.cur so I have pretty high confidence that it's flashing fine. At one point, my printer would beep if I clicked the menu dial, but there was never anything showing on the display. Now I'm not even getting that beep. I am using the stock screen on my CR-10 and I'd really prefer to no have to spend another $50 on a new screen.

I'm at the end of my wits... I have no idea what to try next or how to get this board to work. Is it DoA?

radek8, 2021-01-12:
Do not use a precompiled or foreign bin file.
Marlin has many parameters attached and you will never know what parameters you have uploaded to the printer.
It is always better to compile your own FW

If the BIN file is renamed to CUR, it means that the FW was loaded correctly. It looks like a bad display driver.
What is your display?
Attach a photo of how you are involved,
Do you have configuration files where you can see what configuration you used? Please attach or at least a link from where you downloaded it.

jrkalf, 2021-01-18:
I had the SKR Mini E3 V2 running for 2 months really and today all of a sudden it froze and at reboot it's stuck in a blue screen. I can't find any guide on how to reset the eeprom or reflash the device properly other than using the SDcard. Sadly this is not working. Do I have a bricked device?

radek8, 2021-01-18:
New Marlin flash won't help?

jrkalf, 2021-01-18:
Hi, It was all of a sudden when I warmed up my hotend to 210 celcius to change my PLA. I pulled the pla, then because I needed to grab a roll I wanted to turn off the hotend because it would take me a bit I noticed that the screen was completely unresponsive. I tried switching the device off/on again and it got stuck on a blue screen.

It is also completely unresponsive from Octoprint.
So I made sure to safely switch everything off and make sure that there was no connection to the device (usb cable for octoprint removed)

I've tried downloading the latest firmware from the website, put it on the original SDcard and put it in. Screen remains blue. Even after 5 minutes waiting, no luck. On checking the SDcard in the computer again the firmware.bin wasn't renamed to firmware.cur.

Is there another way to reset/flash the device?

radek8, 2021-01-18:
Yes
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/issues/346

jrkalf, 2021-01-18:
I'll have to figure this one out during the weekend. I have a v2.0 board, I'm making a wild guess the schematics are different. For my lack of proper googling skills, do you happend to have a link to a v2.0 schematic for this as well? I have the USBtty device and the rest of the parts.

radek8, 2021-01-18:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0SCHpdf.PDF

radek8, 2021-01-18:
More people have the same problem
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/224

radek8, 2021-01-18:
https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0/issues/2
Bootloader for your processor?

g2bb, 2021-01-26:
Turns out the issue was with the actual LCD board. After sending back my board and getting a replacement only to have the same issue I ordered a new LCD and it booted up just fine!

closed 2021-01-26 by g2bb

duggyb2005, 2021-12-30:
Just FYI for everyone i was having this issue with my board (V1.2) not even realizing that i was trying to flash V2.0 firmware. DUH! Based on this thread i assumed i Bricked my board and almost ordered a new one. thankfully I kept playing around and grabbed a proper V1.2 firmware and all is good! Just posting this incase it helps someone else!