bigtreetech/BIGTREETECH-SKR-mini-E3 #488: [BUG] Pause/Continue Does not work.

YBeyin, 2021-01-14:
### Description

Hello everyone. I just upgraded my ender 3 pro from v1.1.5 to skr mini e3 v2. I flashed my mainboard firmware with pre compiled bin file. Just tried to use pause/continue but it did not work. So i did compile the firmware by myself but after all pause/continue still does not work. When i click the pause button, printer pauses in few seconds. But when i click the continue, hot end stays exact point, extruder loads 20mm arround filament and a note appears on my tft screen whic says "Paused for user continue?". However im hitting the okay button but still does not continue and keeps asking same question again and again. 

I have Ender 3 Pro with  BL-Touch.
My main board is Skr Mini E3 V2 and my screen is Skr Tft35 E3 V3.
I already updated the screen firmware. When i use the tft's sc card slot, it is works fine but when i use the mainboard's sd card slot this issue appears.

Could you please help me about this situation?
Thanks for advise and tips.
### Steps to reproduce
1. Downloaded latest files from https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3
2. Used Vs Code as correctly.
3. Enabled(from Configration.h) : #define BLTOUCH, #define Z_SAFE_HOMING , #define AUTO_BED_LEVELING_BILINEAR
4. Disabled(from Configration.h) :  #define MIN_SOFTWARE_ENDSTOP_Z
5. Changed(from Configration.h) : #define CUSTOM_MACHINE_NAME "Ender-3" to #define CUSTOM_MACHINE_NAME "Ender-3 Pro YBeyin" ,  #define Z_MIN_PROBE_ENDSTOP_INVERTING false to #define Z_MIN_PROBE_ENDSTOP_INVERTING true ,   #define GRID_MAX_POINTS_X 5 to   #define GRID_MAX_POINTS_X 4.
6. Enabled(from Configration_adv.h) #define BLTOUCH_DELAY 500,   #define BLTOUCH_FORCE_SW_MODE
7. Changed(platformio.ini ) default_envs = STM32F103RC_btt to default_envs = STM32F103RC_btt_512K
8. Compiled successfuly and flashed firmware.bin with sd card from the motherboard's sd card slot.

**Expected behavior**
When i use the pause button , printer pauses.
Then when i use the continue button, printer continues.

**Actual behavior**
When i use the pause button , printer pauses in few seconds.Hot end goes to near of X dimention comes back on to print.
Then when i use the continue button, hot end stays exact point, extruder loads 20mm arround filament and a note appears on my tft screen whic says "Paused for user continue?". However im hitting the okay button but still does not continue and keeps asking same question again and again. Can't even cancel the print untill cut off the energy.

### Additional Information
* 
[Files.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/5816362/Files.zip)

Tetsuo456, 2021-01-17:
Same behaviour here with same hardware. However it has to be related to screen because when I had SKR mini only without the TFT35 E3 V3.0 it worked fine, and I have not changed the firmware of the board since.

YBeyin, 2021-01-18:
Thank you. I am pretty sure it is about the tft card's firmware. I think when i click pause and continue button, tft trying to save the informations to sd card but not using the onboard sd card and informations gone to nowhere because of there is no sd card on the tft sd card slot...

Tetsuo456, 2021-01-19:
Issue https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1523 has been added to the list of known issues https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1218, for your information @YBeyin.

sarvenn, 2021-02-24:
@YBeyin Check the readme section for TFT screens and also check the config.ini file for further prerequisite from marlin side.
I used to had similar problems. I learnt to compile my own marlin firmware and changed my configuration.h and configuration_adv.h files based on my own wished and TFT prerequisites. Those problems are all gone.
Notice that you need to enable nozzle park feature and set the parameter upon your wish.