bigtreetech/BIGTREETECH-SKR-mini-E3 #489: FR [BUG] SKR e3 mini V2.0 on Ender 3 pro

Drewvee2, 2021-01-17:
Hi There,

I've installed the skr e3 mini v2.0 on my ender 3 pro (with Touch-mi probe), and i get many issue ( fan 0 always ON even with the bugfix version, Alarm and reboots...No printer attached......)

I tried to compile my own as I usually do... but nothing change, i can't use my printer since I installed this motherboard...
I've tried marlin bugfix 2.0.0x, 2.0.6.1, 2.0.7.2,........... Hours of compilations and nothing change
Could you please Help me to get a good Firmware and get my printer works ? :)

I get lost... 

**My config is ; Ender 3 pro - SKR E3 MINI V2.0 - TFT35 V3.0 - TOUCH-MI (hotends.fr)**

Thanks for you suggestions ;)

Seelenkind, 2021-02-03:
Change fan0 vs. fan1

dircr, 2021-02-07:
What is your config? I took mine from https://github.com/MarlinFirmware/Configurations/tree/import-2.0.x/config/examples/Creality/Ender-3%20Pro/BigTreeTech%20SKR%20Mini%20E3%202.0, added HOMING_FEEDRATE_Z, compiled with bugfix 2.0.7.2, went fine.

nadavhury, 2021-02-07:
Needed to flash software on the screen also, all errors went away, BLTOUCH was another story, 
Finally made it through a video guide on youtube, I'd recommend adding a manual somewhere in the GitHub repo 'case it's the first result that comes online on google when searching the product.