bigtreetech/BIGTREETECH-SKR-mini-E3 #490: [FR] PTD - Power detection

cookins, 2021-01-18:
hello
i see no explanation anywhere about some port on skr mini e3 v2 board
[image]
can i get some details and i think it needed to publish info about this port.
as i see its pin12? what max voltage allowed? whats NT?

cookins, 2021-01-20:
i found that there place for module call BTT UPS 24v.
so it is something for, at least.
now im curious, if its allowed to use PC12 (pin12?) port and what max voltage is allowed?

dircr, 2021-02-07:
NC = Not Connected?

cookins, 2021-02-07:
already figured - its for this module:
http://ae01.alicdn.com/kf/H33cff675e51c407888889ff323de6c9aO.jpg_q50.jpg

closed 2021-02-07 by cookins