bigtreetech/BIGTREETECH-SKR-mini-E3 #493: [BUG] (SKR Mini E3 V2 Hotend heating up to 275 at all times)

8qBITs, 2021-01-22:
### Description

### Steps to reproduce

1. [Turn machine on]

**Expected behavior**

**Actual behavior**

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.

hitokiri8x, 2021-01-28:
Are you sure to have wired up the hotend to the correct input?
At first I wired up the hotend to 24v fixed output( which the hotend fan should be connected )

jlmeredith, 2021-06-04:
I had the same issue and determined that the thermistor port was blown on the board. Apparently there is not any protection on this port. My guess would be that I had a static shock through the extruder and killed the port. Seems that in the newest version of the board they have added this (STM32F103RET6 vs the STM32F103RCT6) but not 100% sure.