bigtreetech/BIGTREETECH-SKR-mini-E3 #494: [FR] Pre-build Firmwares to be Updated when Marlin & BTT SKR Configs are Updated

sarvenn, 2021-01-24:
**Is your feature request related to a problem? Please describe.**
Pre-build binaries are too old. SKR board config files and marlin base firmware (bugfix 2.xxx) had lots of updates, but pre-build firmwares in this repo are not  up to date.
I know, many of you may suggest me to build my own fw using vscode but since BTT products are end user items, it's firmware also should be up to date and hassle free to install for the brand reputation. Also we should consider people are not familiar with compiling a fw, changing the config files etc.
There should be more firmware complexties like direct drive users, more grid points for probing etc.
Marlin mode user configurable menus also should be activated as much as possible in config files and pre-build firmwares.

**Describe the solution you'd like**
Touch screen firmware pre-build firmwares are updated automatically when the master files are updated/merged by a PR.
Just like it, BTT SKR mini E3 pre-build firmwares also should be uptated automatically when the Marlin fw repository and/or Marlin SKR mini E3 config files are updated.
E.g.:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1450
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1463

I had asked for help at Facebook group however could not get any help. So I had to open a new ticket here.

radek8, 2021-01-24:
Marlin is variable and has too many parameters. If 100 people come to download FM for the record, everyone will require different parameters. It's better to create an FW yourself than to rely on someone creating it for me with the parameters I need.
It is impossible to satisfy everyone.

sarvenn, 2021-01-25:
> Marlin is variable and has too many parameters. If 100 people come to download FM for the record, everyone will require different parameters. It's better to create an FW yourself than to rely on someone creating it for me with the parameters I need.
> It is impossible to satisfy everyone.

I agree that it not possible to satisfy everyone. But basic requirements are almost common for everyone and anyone.
Basically, what I suggest is that firmware can be updated automatically when the configuration files are updated in Marlin repository. Advanced users are already capable of compiling their own firmware based upon their needs. But beginners may require to have a precompiled firmware. I think that's why there are already precompiled firmwares in this repository. What I suggest is to keep it up to date when a update is done in Marlin SKR mini specific configurations or base marlin fw. It is already being done in touch screen repository. When someone merges a PR into master, precompiled firmwares are also updated.

For example when I first bought my SKR mini E3 V2 and TFT35 V3, I found this repository and read the readme section, then I successfully updated my firmwares from precompiled repositories. Now I am capable of compiling my own touchscreen fw and edit my config.ini file. However I still have road to walk to compile my motherboard firmware. I still rely on advanced user's compiled firmwares, because I trust them that they have done it correctly. I am watching and learning about the Marlin features and their benefits at the moment.

radek8, 2021-01-25:
Marlin has its own code repository. HW from Bigtreetech has its. I acknowledge that they are poorly maintained by Bigtreetech and I do not want to stand up for them. But it is not possible to set BIN files to be generated automatically after each update.

You can't compare it to a touch screen FW that's written directly for several specific types of displays. In addition, it is designed so that it can be configured with an INI file after compilation. This is not possible with Marlin. The resulting code would be so large that it would not fit in any board's memory.
Compare the size of a BIN file for a TFT display and the simplest BIN file with Marin.
They're the same size, but Marlin controls the entire printer and calculates movement routes. TFT is only an interface that sends commands that you enter via the display or SD card.

If someone should manually generate new BIN files after each change (Marlin changes every day :-) it would be too much time. Imagine that you have 20 boards in your portfolio, you will create 6 "basic" configurations for each.
This is 20 * 6 = 120 compilations that maybe no one will download because they don't suit him.
Updating the precompiled Marlin every 3-4 months seems adequate to me.

sarvenn, 2021-01-26:
Hi @radek8,
Thanks for the brief explanation. I thought it was simple as it is like TFT bin files are updated automatically. Now I understood the complexity of my request and the roadblocks in front of it.
On the other hand your last proposal sounds really good. (updating precompiled files every 3-4 months).
So, if there is nothing else to do, I can close the ticket. Any comments?

sarvenn, 2021-01-29:
@thisiskeithb , @bigtreetech any comments?

sarvenn, 2021-02-24:
No comments have been received and issue is closed.

closed 2021-02-24 by sarvenn