bigtreetech/BIGTREETECH-SKR-mini-E3 #496: BLTouch ABL is all wrong

Zane155, 2021-01-25:
I have an Ender 3 with an SKR mini E3 V 1.2  and I recently have installed a BLTouch onto my printer. The BLTouch installed fine and I flashed the new firmware, however now when I try to auto bed level the printer rises about an inch above the printer bed and tries to auto bed level from up there. The BLTouch releases the probe and the Z-Axis moves down, but it only moves down a few millimeters than goes back up. My screen says-Setting Stored (722 byt es; crc 33304) but the BLTouch never even touch the build plate at all and after I attempt an ABL I cannot move the Z-Axis any lower than what it thinks zero is. I am very new to 3D printing and I have no idea what I am doing so any help would be greatly appreciated.

Edorka, 2021-02-02:
Opposite happens too, even contacting a surface ABL turns red but _WONT_ stop so continues lowering down, even displacing the bed.

Edorka, 2021-02-03:
> I have an Ender 3 with an SKR mini E3 V 1.2 and I recently have installed a BLTouch onto my printer. The BLTouch installed fine and I flashed the new firmware, however now when I try to auto bed [...]

Zane check if your old z-stop is still conected and it's position makes it to activate before the probe can reach the surface

YoAdriaaannn, 2021-02-22:
Enable baby stepping in your firmware.