bigtreetech/BIGTREETECH-SKR-mini-E3 #497: [HELP] TMC2209 Stepper Motor Configuration

VGaborHu, 2021-01-27:
Hello, 

I'm building my own 3d printer. and i need a little help how can i calculate the Stepper Motor Current value. 

I ordered SKR Mini E3 V2.0 Board, and Nema17 Stepper Motor 17HS4401 stepper motors. 

[image]

#if HAS_TRINAMIC_CONFIG

  #define HOLD_MULTIPLIER    0.5  // Scales down the holding current from run current
  #define INTERPOLATE       true  // Interpolate X/Y/Z_MICROSTEPS to 256

  #if AXIS_IS_TMC(X)
    #define X_CURRENT       800        // (mA) RMS current. Multiply by 1.414 for peak current.
    #define X_CURRENT_HOME  X_CURRENT  // (mA) RMS current for sensorless homing
    #define X_MICROSTEPS     16    // 0..256
    #define X_RSENSE          0.11
    #define X_CHAIN_POS      -1    // <=0 : Not chained. 1 : MCU MOSI connected. 2 : Next in chain, ...
  #endif

How can i calculate the X_CURRENT ?

Thanks!

siochs, 2021-02-01:
This might helps you: https://mak3r.de/2020/04/03/stepper-driver-current-and-vref-calculator/

Seelenkind, 2021-02-03:
> How can i calculate the X_CURRENT ?

max 50% of rated current is good. 
1500mA x 50% = 750 mA

The higher the current, the better the motors run. But also get warmer.
The lower the current, the cooler the motors run. But can lose steps.
In practice this means setting as much as necessary and as little as possible.