bigtreetech/BIGTREETECH-SKR-mini-E3 #499: [BUG] No USB connection

dedepene, 2021-01-28:
### Description

Can't connect to the printer over USB
### Steps to reproduce

1. power ender 3 on
2. plug in usb cable
3. pc, both windows 10 and debian 10 do not recognized a USB connected device

**Expected behavior**

opening a COM port
**Actual behavior**

nothing
### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.
https://pastebin.com/EYpFU7HZ

wprojects, 2021-02-02:
Are you using Cura? Your issue might be a layer deeper than this but it's worth a shot!

Unplug your printer from the computer and then do the following.

Locate the Ultimaker Cura 4.x folder, often found at C:\Program Files

Navigate to the sub directory Plugins\USBPrinting

Open the file AutoDetectBaudJob.py using Notepad or another text editor

Comment out the following lines by placing a # symbol in front of each (as shown below).

# try:
# programmer.connect(self._serial_port)
# serial = programmer.leaveISP()
# except ispBase.IspError:
# programmer.close()
 Save the changes to the file and restart the Cura software. This should resolve the problem with USB connections on the SKR Mini E3 board.

Good luck!

dedepene, 2021-02-02:
> Are you using Cura? Your issue might be a layer deeper than this but it's worth a shot!
> 
> Unplug your printer from the computer and then do the following.
> 
> Locate the Ultimaker Cura 4.x folder, often found at C:\Program Files
> 
> Navigate to the sub directory Plugins\USBPrinting
> 
> Open the file AutoDetectBaudJob.py using Notepad or another text editor
> 
> Comment out the following lines by placing a # symbol in front of each (as shown below).
> 
> l # try:
> l # programmer.connect(self._serial_port)
> l # serial = programmer.leaveISP()
> l # except ispBase.IspError:
> l # programmer.close()
> ignore the l infront of the #, I had to put those infront of the # so that github didn't format it!
> 
> Save the changes to the file and restart the Cura software. This should resolve the problem with USB connections on the SKR Mini E3 board.
> 
> Good luck!

Using Simplify3D instead of Cura. Any pointers here?

CrackerA, 2021-03-14:
I have just upgraded to SKR e3 mini v1.2.

The usb on this board is dead. I have compiled firmware for serial ports as follows,

#define SERIAL_PORT -1
#define SERIAL_PORT_2 2

#define SERIAL_PORT 2
#define SERIAL_PORT_2 -1

#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1 This would not finish build

Nothing from pi using octoprint. Raspberry pi has no communication with printer (SKR board.
PC nothing.

Have tried the above with cura with no success.

CrackerA, 2021-03-20:
For anyone with the above problem:

On putting the SKR board back in I found the following.
Check that there is a gap of 1mm or so around the USB port on the ender 3 pro. I found that my USB was just touching the ender 3 pro case. This was the cause of no communication with the pi. Filed the hole in the case and taped the edges. Now all working.